pothosware/SoapySDR #260: HPSDR support for SoapySDR?

frohro, 2020-05-04:
Hi All,
I would like to use some of my hpsdr hardware with software that interfaces with SoapySDR.  There seem to be several ways of pursuing this, and I'm hoping for some advice.  Here are the ones I've thought of.  They may be combined in some cases. 

- Linhpsdr has some code available so that it will interface with SoapySDR.  This is really going the wrong way for me, because I want to interface hpsdr hardware with SoapySDR using software, but it could be useful.
- There is a [gnuradio module for hpsdr](https://github.com/Tom-McDermott/gr-hpsdr) hardware.  Perhaps I could just pipe data from one to the other.  I'm afraid that I might not have all the functionality that way, and that it will require gnuradio and may be a little clumsy, but it might be the quick and dirty way.
- Write my own SoapySDR plugin that translates from hpsdr to SoapySDR.

What do you recommend?

Thanks,

Rob

guruofquality, 2020-05-04:
> Write my own SoapySDR plugin that translates from hpsdr to SoapySDR.

I would do this. Its going to be the cleanest most likely to work straightforward option.
Take a look at the driver guide: https://github.com/pothosware/SoapySDR/wiki/DriverGuide

I wouldnt wrap the gr-hpsdr project per-say but it looks to have an api "HermesProxy.h" and some decent examples of both settings and streaming in "hermesNB_impl.cc". So you might have an easier time doing this than it appears by re-using this code, just not exactly in the form its in. You know, just mind the copyright and the GPL in this case

As always, I offer hosting on this organization if a support modules is something you want to pursue. 

Best!

frohro, 2020-05-05:
> CAUTION: This email originated from outside the Walla Walla University
> email system.

Thanks Josh,

Is there a simple example available or only the skeleton?

Thanks,

Rob

guruofquality, 2020-05-05:
RTL SDR is a pretty compact example: https://github.com/pothosware/SoapyRTLSDR

cjheath, 2022-09-05:
It's not in C, but for an example, you could look at the connector code in Go for openwebrx: github.com/jancona/hpsdrconnector

ncorgan, 2022-12-26:
If I can have some hardware loaned, I can take this on.