pothosware/SoapySDR #261: Python Bindings not Returning Devices

deltatango2, 2020-05-05:
Under the SoapySDR python bindings, a call to:

`results = SoapySDR.Device.enumerate()`

returns empty

However on the same machine, a call in Cpp to:
`SoapySDR::KwargsList results = SoapySDR::Device::enumerate();`

returns

Found device #0: driver = sdrplay
label = SDRplay Dev0 RSPdx 

Similarly, SoapySDRUtil finds the device as well.  Any pointers?

guruofquality, 2020-05-05:
I would check for multiple versions of the SoapySDR library: https://github.com/pothosware/SoapySDR/wiki/ConfigGuide#avoid-simultaneous-installs

deltatango2, 2020-05-05:
Perfect thanks!  Checking the last line in `make install` output showed where it was placing the `SoapySDR.py` file.  After appending that path to my python file all is well.

Thanks again for the lightning fast reply.

closed 2020-05-05 by deltatango2