pothosware/SoapySDR #263: Virtual device from application support?

Aang23, 2020-05-15:
This is more of a random idea that would be very convenient for what I think are very rare usecases but here it is nonetheless!

I am working on several projects where I am continuously listening to a pretty wide bandwidth to automatically collect data, as an example on the 137Mhz satellite band. On such a setup only 1 SDR really need to be used to cover everything of interest (~1Mhz covers it all).
But currently, if my SDR is being used continuously by my program, as expected, using it on another program is impossible. 
An easy fix it getting a second SDR and splitter, but it would be much more convenient if it was possible to register a SoapySDR device, streaming the samples that are whatever happens, already sampled by the application, for use by another Soapy-supported program (or perhaps even allow multiples users at once, since SDR settings wouldn't be available?).
 
My thought on getting this working is a custom driver, restricting frequency, gain and sample rate controls, but this requires providing a separate driver along with a way to feed the samples in.

I have no idea what it would take to support this kind of stuff in Soapy, and again, this is more of an idea than anything else, since I highly doubt most uses would ever need this.

guruofquality, 2020-05-16:
I think the way to do it that would not end up changing various device drivers would be to support a mode in SoapyRemote so the network stream could be sent to multiple destination sockets. Its really the only way to effectively share a device amongst multiple processes, hosts, etc.

So the underlying driver wouldn't need to change, but the SoapyRemote server would support opening a stream on the same channels and I guess maintaining a list of sockets for the forwarding thread instead of the one.

Something like that...

* https://github.com/pothosware/SoapyRemote

pe1chl, 2021-01-24:
I have had this requirement multiple times and it always surprised me that all SDR applications and drivers lack this capability.
So it would be very nice when there would be support for this in a central point of the system like SoapySDR.
A solution via something like SoapyRemote would be acceptable as long as it does not cause an unreasonable system load when used only locally.
Like Aang23 I would probably prefer a "hack" that just sniffs what a device is already sending to one application and make the same samples available to another application (which would have no effective control over the device).

zuckschwerdt, 2021-01-24:
From the applications point of view this driver would look significantly different. You can't assume any (default) settings and need to account for changes. Without a mechanism for "setting events" the application would need to poll relevant settings with each frame. E.g. knowing the sample rate is critical to use any data. This needs to be addressed in applications. Perhaps with a driver that forces applications to think about that model of operation and opt-in explicitly.
Hacking this might be of limited use with a user experience of "things not quite right" most of the time.

pe1chl, 2021-01-24:
Yes sure.  But this is kind of a "niche application" which assumes users that are aware of what they are doing.  In an environment like what AAng23 described (and what I would do as well), the settings are rarely changed if at all.
I think it would be OK when the 2nd+ application (the ones without control) can have the usual settings interface but the available options are limited to what the main application has already set, and their settings changes are just ignored.  They receive the data as sent to the first application.  It is the responsibility of the operator to set all applications to the same settings so that it "magically works".

zuckschwerdt, 2021-01-24:
> so that it "magically works"

That's the part I'm worried about, with the usual applications it would very likely "magically **not** work". (Just to make sure, when you talk about "niche application" you mean "niche use-case", right? With "applications" I mean the user software.) All that software needs provisions to handle the case where settings (and commands to e.g. start and stop a driver) basically go the "reverse" direction: the driver sets the parameters in the application -- control flow in common applications currently doesn't handle that well.

Maybe to get this started and check the idea out, perhaps use rtl_tcp -- it has a dumb simple approach and can be implemented (server or client) with little code. Put an rtl_tcp server in front of Soapy, then test some rtl_tcp enabled application to see if things can be handled. Beware that rtl_tcp has no way to acknowledge or query settings -- on failure the server should simply close the connection. This will throw applications in a loop if they assume they can apply settings, otoh just ignoring issued settings will lead to an invalid state in the application. I might be up to provide some quick example code for that, but I don't see any application that would handle this.
What applications do you have in mind and do they have robust rtl_tcp support?

pe1chl, 2021-01-24:
Indeed with "nice application" I mean "niche use case".  Sorry, English is not my first language.
For rtl_tcp this already exists!  E.g. https://github.com/alexander-sholohov/rtlmuxer (others exist too).
Of course the use is limited to cases where all applications can live with the same settings (a problem, in practice, but outside the scope of this thread).
I would e.g. like to use gqrx on an SDRplay RSP1a via soapy, and at the same time process the sample data from the RSP1a in another program to look for a reference signal in the same passband, calculate its frequency, and set a correct value of LNB_LO in grqx via its API.

starvald, 2021-02-06:
there is https://github.com/petrinm/SoapyShared but i've never tried it.

would be a handy feature for soapysdr to have.

pe1chl, 2021-02-06:
Thanks!  That is what I was looking for...  indeed it would be nice if such a thing could be part of soapysdr, but for now I have something I can try aside from it.

closed 2022-03-25 by zuckschwerdt