pothosware/SoapySDR #264: Corrupted call stack trying to change sample rate.

agrecascino, 2020-05-27:
Calling into SoapyLMS7 trashes my call stack going into SetSampleRate, calling the wrong function and segfaulting with a corrupt stack. I know this _probably_ isn't your problem, but I was hoping for some guidance on where to go from here.

```

> Thread 1 "rtl_airband" hit Breakpoint 1, SoapySDRDevice_setSampleRate (device=0x561b8, 
>     direction=1, channel=0, rate=20000000)
>     at /home/pi/airband-radio/install/radio/SoapySDR/lib/DeviceC.cpp:564
> 564	    __SOAPY_SDR_C_TRY
> (gdb) bt
> #0  SoapySDRDevice_setSampleRate (device=0x561b8, direction=1, channel=0, rate=20000000)
>     at /home/pi/airband-radio/install/radio/SoapySDR/lib/DeviceC.cpp:564
> #1  0x00022134 in soapysdr_init (input=0x45250) at input-soapysdr.cpp:242
> #2  0x00019318 in input_init (input=0x45250) at input-common.cpp:61
> #3  0x00017c90 in main (argc=3, argv=0xbefff634) at rtl_airband.cpp:923
> (gdb) step
> SoapySDRDevice_clearError ()
>     at /home/pi/airband-radio/install/radio/SoapySDR/lib/DeviceC.cpp:564
> 564	    __SOAPY_SDR_C_TRY
> (gdb) step
> SoapySDRDevice_setSampleRate (device=0x561b8, direction=1, channel=0, rate=20000000)
>     at /home/pi/airband-radio/install/radio/SoapySDR/lib/DeviceC.cpp:28
> 28	    lastErrorStatus = 0;
> (gdb) step
> 565	    device->setSampleRate(direction, channel, rate);
> (gdb) step
> SoapyLMS7::listSampleRates (this=0x1, direction=0, channel=<optimized out>)
>     at /home/pi/airband-radio/install/radio/LimeSuite/SoapyLMS7/Settings.cpp:479
> 479	    if (LMS_GetSampleRateRange(lms7Device, direction == SOAPY_SDR_RX, &range))
> (gdb) bt
> #0  SoapyLMS7::listSampleRates (this=0x1, direction=0, channel=<optimized out>)
>     at /home/pi/airband-radio/install/radio/LimeSuite/SoapyLMS7/Settings.cpp:479
> #1  0xb6c5a974 in SoapySDRDevice_setSampleRate (device=0x561b8, direction=1, channel=0, 
>     rate=20000000) at /home/pi/airband-radio/install/radio/SoapySDR/lib/DeviceC.cpp:565
> #2  0x00022134 in soapysdr_init (input=0x45250) at input-soapysdr.cpp:242
> #3  0x00019318 in input_init (input=0x45250) at input-common.cpp:61
> #4  0x00017c90 in main (argc=3, argv=0xbefff634) at rtl_airband.cpp:923

```

guruofquality, 2020-05-29:
It kind of looks like an API issue, make sure you didnt have like versions of the development libraries for SoapySDR installed or something like that https://github.com/pothosware/SoapySDR/wiki/ConfigGuide#avoid-simultaneous-installs

agrecascino, 2020-06-04:
Yeah that was it, closing.

closed 2020-06-04 by agrecascino