pothosware/SoapySDR #265: Can't run my app in debug mode on windows when using SoapySDR.

AlexandreRouma, 2020-06-22:
Hi
I'm having issues running my app in debug mode. When I do, I'm getting a memory access violation from inside SoapySDR when running SoapySDR::Device::Enumerate.

The exact error is :
`Unhandled exception at 0x00007FFE85AA34A2 (SoapySDR.dll) in sdrpp.exe: 0xC0000005: Access violation reading location 0x0000000000000019.`

Any idea what the issue could be?

Thanks in advance.

guruofquality, 2020-06-22:
With C++ in MSVC, you have to compile all of the dependent libraries with the same build flags or you get ABI incompatibilities.

AlexandreRouma, 2020-06-22:
Ah, that's a problem :/
Having to build everything in debug is a pain. Especially when having to make sure it doesn't try to load the stuff that's on the PATH var.
Guess I'll just have to debug on linux lol

guruofquality, 2020-06-22:
At least when I have had to debug on windows, the CMAKE_BUILD_TYPE "RelWithDebInfo" has given me reasonably good tracebacks. Its basically release mode with some extra debug info, not sure on the details. So all of the windows packages I release have that mode set when possible. If thats worth anything

AlexandreRouma, 2020-06-22:
I'll try that, thanks!

closed 2021-06-06 by guruofquality