pothosware/SoapySDR #267: error: too many arguments to function ‘SoapySDRDevice_setupStream’ in

MountainLogic, 2020-07-06:
This may be related to https://github.com/pothosware/SoapySDR/issues/230 

My goal is to build build this tutorial https://myriadrf.org/news/limesdr-made-simple-part-8-c-examples-soapysdr-api/ which uses [C_API_Example ](https://github.com/pothosware/SoapySDR/wiki/C_API_Example) as a starting point.  When I try to compile the code as provided in the repo I am getting a  SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, SOAPY_SDR_CF32, NULL, 0, NULL) function signature mismatch.  The argument lists in the example and Device,h file are clearly of different length.

Build environment:
```
I am working under the windows linux subsystem with Ubuntu 
Distributor ID: Ubuntu                                                                                                                                                                                               Description:    Ubuntu 18.04.4 LTS                                                                                                                                                                                   Release:        18.04                                                                                                                                                                                                Codename:       bionic   
```     

I followed the install instructions https://github.com/pothosware/PothosCore/wiki/Ubuntu and then built lime driver following the directions in Eric's answer here: https://stackoverflow.com/questions/55850748/how-to-properly-install-soapysdr-with-limesdr-usb 

It also seems that directions or installing Lime are should be added to the Ubuntu install guide.

```

> Executing task: /usr/bin/gcc -g /home/foo/wip/lime/example.c -o /home/foo/wip/lime/example <

/home/foo/wip/lime/example.c: In function ‘main’:
/home/foo/wip/lime/example.c:71:37: warning: passing argument 2 of ‘SoapySDRDevice_setupStream’ makes integer from pointer without a cast [-Wint-conversion]
     SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, SOAPY_SDR_CF32, NULL, 0, NULL);
                                     ^
In file included from /home/foo/wip/lime/example.c:1:0:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘int’ but argument is of type ‘SoapySDRStream ** {aka struct SoapySDRStream **}’
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/foo/wip/lime/example.c:71:48: warning: passing argument 3 of ‘SoapySDRDevice_setupStream’ makes pointer from integer without a cast [-Wint-conversion]
     SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, SOAPY_SDR_CF32, NULL, 0, NULL);
                                                ^~~~~~~~~~~~
In file included from /home/foo/wip/lime/example.c:1:0:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const char *’ but argument is of type ‘int’
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/foo/wip/lime/example.c:71:62: warning: passing argument 4 of ‘SoapySDRDevice_setupStream’ from incompatible pointer type [-Wincompatible-pointer-types]
     SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, SOAPY_SDR_CF32, NULL, 0, NULL);
                                                              ^~~~~~~~~~~~~~
In file included from /home/foo/wip/lime/example.c:1:0:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const size_t * {aka const long unsigned int *}’ but argument is of type ‘char *’
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/foo/wip/lime/example.c:71:78: warning: passing argument 5 of ‘SoapySDRDevice_setupStream’ makes integer from pointer without a cast [-Wint-conversion]
     SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, SOAPY_SDR_CF32, NULL, 0, NULL);
                                                                              ^~~~
In file included from /home/foo/wip/lime/example.c:1:0:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘size_t {aka const long unsigned int}’ but argument is of type ‘void *’
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/foo/wip/lime/example.c:71:5: error: too many arguments to function ‘SoapySDRDevice_setupStream’
     SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, SOAPY_SDR_CF32, NULL, 0, NULL);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/foo/wip/lime/example.c:1:0:
/usr/local/include/SoapySDR/Device.h:295:31: note: declared here
 SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
The terminal process terminated with exit code: 1
```

Amazing work on Soapy.  Really appreciate the effort.

guruofquality, 2020-07-06:
The recent release changed the API for consistency to return the stream pointer, you can make the change or if you need version compatibility, here is an example in rx tools: https://github.com/rxseger/rx_tools/blob/master/src/convenience/convenience.c#L500

MountainLogic, 2020-07-06:
Thanks, Josh, Nice example and it looks like a good starting point for learning the API.

closed 2020-07-06 by MountainLogic