pothosware/SoapySDR #268: Auto gain gives much stronger signal than manual

dvdesolve, 2020-07-14:
We use SoapySDR in our [glrpt](https://github.com/dvdesolve/glrpt) software. There is one observation that for at least RTL SDR dongles (and for Airspy too, IIRC) using Auto gain mode gives much stronger signal (approx 3 times) when compared to the dongle's maximum available gain (around 49 dB according to the SoapySDR info). It looks somewhat wrong and we don't know what can cause such behavior.

zuckschwerdt, 2020-07-14:
Gains for the RTL-SDR are a complicated topic. There are different tuners with a varying setup of gain stages. If you have the rather common R820T the manual gain setting calls down to https://github.com/osmocom/rtl-sdr/blob/master/src/tuner_r82xx.c#L953 and you might find the culprit in the fixed VGA gain. I guess it was choosen with good reason and after much experimentation, but if you are adventurous you could try to play with that.

dvdesolve, 2020-07-14:
Yes, I have R820T2 tuner

zuckschwerdt, 2020-07-14:
Looks like that. And 10 dB is a factor of about 3. Of course somebody already tried that https://www.rtl-sdr.com/new-r820t-driver-lnamixervga-gain-settings/ which might give additional insight.

dvdesolve, 2020-07-14:
Thank you a lot! Things are much clearer now!

closed 2020-07-14 by dvdesolve