pothosware/SoapySDR #270: Afedri SDR-Net support?

gvanem, 2020-08-04:
I had just purchased a 2nd hand [Afedri SDR-Net](https://www.afedri-sdr.com/index.php) box. This box has a USB-connector *and* an Ethernet connector.

I'm tinkering with the idea to implement a Soapy-driver for this Afedri box myself, but w/o really having a clue where to
start. But using [SoapyNetSDR](https://github.com/pothosware/SoapyNetSDR) as a boiler-plate is perhaps a start?

Or perhaps creating a local shim-process that *fakes* the NetSDR protocol is better?

Afedri also has a special network-discovery protocol described in `sdr_disovery.cpp` in the .zip at
[SDR Network Control Box x4 v5.21h - source code](https://www.afedri-sdr.com/index.php/downloads/category/21-sdr-control-source-code).

I'm not sure how that fits in with a `SoapySDR::KwargsList find_afedriSDR(const SoapySDR::Kwargs &args)`.

guruofquality, 2020-08-05:
SoapyNetSDR can be a good boilerplate, especially because it manually implements some socket code. There is also a driver guide: https://github.com/pothosware/SoapySDR/wiki/DriverGuide

In regards to discovery, ideally there would be a way to discover devices on the network. And to return something identifiable for each device, like an address or serial number.

If not, you can do something like red pitaya, where the find function only discovers the device that is explicitly identified

* https://github.com/pothosware/SoapyRedPitaya/blob/master/SoapyRedPitaya.cpp#L527

gvanem, 2020-08-05:
> In regards to discovery, ideally there would be a way to discover devices on the network. And to return something 
> identifiable for each device, like an address or serial number.

And this could or should be cached in a file (or Windows Registry)? With minimal testing so far, this Afedri discovery protocol is not very reliable AFAICS.

I will look into how RedPitaya is coded.

gvanem, 2020-08-07:
After creating a small test program for Afredi-SDR, it seems it's very similar to `RFSPACE_SDR_IQ` etc.
So it would be much easier to extend *SoapyNetSDR* and [`radio_type`](https://github.com/pothosware/SoapyNetSDR/blob/master/SoapyNetSDR.hpp#L201) with a `AFREDI_SDR_NET` value.

gvanem, 2020-08-07:
Closing as per above comment.

closed 2020-08-07 by gvanem

2E0WKS, 2022-02-06:
https://www.moetronix.com/files/NetSdrInterfaceSpec105.pdf

nmaster2042, 2024-02-11:
There is a new Soapy driver for AFEDRI SDR.

See here: https://github.com/alexander-sholohov/SoapyAfedri