pothosware/SoapySDR #272: cannot cmake SoapySDR for Python3 on Debian 10

mariocannistra, 2020-08-26:
Hello.
I'm working on  Debian 10 Buster  and using  Python 3.7.3
Also using  update-alternatives  to be able to switch.  Current config:

```
mario@debian:~/SoapySDR/build$ update-alternatives --config python
Sono disponibili 2 scelte per l'alternativa python (che fornisce /usr/bin/python).

  Selezione    Percorso            Priorità  Stato
------------------------------------------------------------
* 0            /usr/bin/python3.7   2         modalità automatica
  1            /usr/bin/python2.7   1         modalità manuale
  2            /usr/bin/python3.7   2         modalità manuale

```

When I try to cmake SoapySDR:

`mario@debian:~/SoapySDR/build$  cmake ../ -DCMAKE_BUILD_TYPE=Release -DENABLE_PYTHON3=ON`

I get the following output where "**Python3, python3 bindings**" is listed in the **disabled features**:

> -- 
> -- #############################################
> -- ## Begin configuration for Python support...
> -- #############################################
> -- Enabling optional Python bindings if possible...
> -- SWIG_FOUND: TRUE - 3.0.12
> -- PYTHONINTERP_FOUND: TRUE - 3.7.3
> -- PYTHON_EXECUTABLE: /usr/bin/python
> -- PYTHON_INSTALL_DIR: ${prefix}/lib/python3.7/dist-packages
> -- PYTHONLIBS_FOUND: TRUE - 3.7.3
> -- PYTHON_INCLUDE_DIRS: /usr/include/python2.7;/usr/include/x86_64-linux-gnu/python2.7
> -- PYTHON_LIBRARIES: -L/usr/lib/python2.7/config-x86_64-linux-gnu -L/usr/lib -lpython2.7 -lpthread -ldl  -lutil -lm  -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions
> -- CMAKE_SWIG_FLAGS=-c++;-threads;-I/home/mario/SoapySDR/include
> CMake Warning (dev) at /usr/share/cmake-3.13/Modules/UseSWIG.cmake:564 (message):
>   Policy CMP0078 is not set.  Run "cmake --help-policy CMP0078" for policy
>   details.  Use the cmake_policy command to set the policy and suppress this
>   warning.
> Call Stack (most recent call first):
>   python/CMakeLists.txt:155 (SWIG_ADD_LIBRARY)
> This warning is for project developers.  Use -Wno-dev to suppress it.
> 
> -- 
> -- #############################################
> -- ## Begin configuration for Python3 support...
> -- #############################################
> -- Enabling optional Python3 bindings if possible...
> -- SWIG_FOUND: TRUE - 3.0.12
> -- PYTHON3INTERP_FOUND: TRUE
> -- PYTHON3_EXECUTABLE: /usr/bin/python3
> -- PYTHON3_INSTALL_DIR: ${prefix}//usr/local/lib/python3/dist-packages
> -- PYTHON3LIBS_FOUND: TRUE
> -- PYTHON3_INCLUDE_DIRS: /usr/include/python3.7m;/usr/include/python3.7m
> -- PYTHON3_LIBRARIES: -L/usr/lib/python3.7/config-3.7m-x86_64-linux-gnu -L/usr/lib -lpython3.7m -lcrypt -lpthread -ldl  -lutil -lm  -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions
> -- 
> -- ######################################################
> -- ## SoapySDR enabled features
> -- ######################################################
> -- 
>  * Library, runtime library v0.8.0-g580b94f3
>  * Apps, command line applications
>  * Tests, library unit tests
>  * Docs, doxygen documentation
>  * Python, python bindings v3.7.3
> 
> -- ######################################################
> -- ## SoapySDR disabled features
> -- ######################################################
> -- 
>  * Python3, python3 bindings
> 
> -- SoapySDR version: v0.8.0-g580b94f3
> -- ABI/so version: v0.8
> -- Install prefix: /usr/local
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/mario/SoapySDR/build

**I can actually build the code without erros but when I try to use SoapySDR from Python3 I'm getting this error:**

> Traceback (most recent call last):
>   File "/usr/local/lib/python3.7/dist-packages/SoapySDR.py", line 28, in swig_import_helper
>     return importlib.import_module(mname)
>   File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 670, in _load_unlocked
>   File "<frozen importlib._bootstrap>", line 583, in module_from_spec
>   File "<frozen importlib._bootstrap_external>", line 1043, in create_module
>   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
> ImportError: dynamic module does not define module export function (PyInit__SoapySDR)
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "pwrlogpi.py", line 3, in <module>
>     import SoapySDR
>   File "/usr/local/lib/python3.7/dist-packages/SoapySDR.py", line 31, in <module>
>     _SoapySDR = swig_import_helper()
>   File "/usr/local/lib/python3.7/dist-packages/SoapySDR.py", line 30, in swig_import_helper
>     return importlib.import_module('_SoapySDR')
>   File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
> ImportError: dynamic module does not define module export function (PyInit__SoapySDR)

Could you please advise on how to proceed?
Thank you,
Mario

guruofquality, 2020-08-29:
```
-- PYTHON_EXECUTABLE: /usr/bin/python
-- PYTHON_INSTALL_DIR: ${prefix}/lib/python3.7/dist-packages
-- PYTHONLIBS_FOUND: TRUE - 3.7.3
-- PYTHON_INCLUDE_DIRS: /usr/include/python2.7;/usr/include/x86_64-linux-gnu/python2.7
-- PYTHON_LIBRARIES: -L/usr/lib/python2.7/config-x86_64-linux-gnu -L/usr/lib -lpython2.7 -lpthread -ldl -lutil -lm -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions
```

The build configuration is very messed up. From the looks of it your /usr/bin/python is a python3 interpreter but the flags come from /usr/bin/python-config which is python2. So I dont think the system's executables make sense.

> -DENABLE_PYTHON3=ON

FYI you dont need this, the first python configuration can build against python3 as well if configured right. The reason for both is those dual python2 and 3 systems, so the build can do both.

MartinPolytech, 2021-01-28:
Hello ! :) 

We think we've got the same issue : 

 54%] Built target SoapySDR
[ 64%] Built target SoapySDRUtil
[ 69%] Built target TestFormatParser
[ 73%] Built target TestTimeConversion
[ 78%] Built target TestKwargsMarkup
[ 83%] Built target TestConvertTypes
[ 85%] Built target docs
[ 92%] Built target _SoapySDR
[100%] Built target _SoapySDR3
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/lib/python3/dist-packages/share/cmake/SoapySDR/SoapySDRConfig.cmake
-- Up-to-date: /usr/local/lib/python3/dist-packages/share/cmake/SoapySDR/SoapySDRUtil.cmake
-- Up-to-date: /usr/local/lib/python3/dist-packages/share/cmake/SoapySDR/SoapySDRConfigVersion.cmake
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Errors.hpp
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Modules.hpp
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Types.h
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Config.hpp
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Logger.hpp
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Time.hpp
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Types.hpp
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/ConverterRegistry.hpp
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/ConverterPrimitives.hpp
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Formats.h
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Device.hpp
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Constants.h
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Registry.hpp
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Errors.h
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Converters.h
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Version.hpp
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Formats.hpp
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Modules.h
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Version.h
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Device.h
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Time.h
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Logger.h
-- Up-to-date: /usr/local/lib/python3/dist-packages/include/SoapySDR/Config.h
-- Up-to-date: /usr/local/lib/python3/dist-packages/lib/libSoapySDR.so.0.8.0
-- Up-to-date: /usr/local/lib/python3/dist-packages/lib/libSoapySDR.so.0.8
-- Up-to-date: /usr/local/lib/python3/dist-packages/lib/libSoapySDR.so
-- Up-to-date: /usr/local/lib/python3/dist-packages/share/cmake/SoapySDR/SoapySDRExport.cmake
-- Up-to-date: /usr/local/lib/python3/dist-packages/share/cmake/SoapySDR/SoapySDRExport-release.cmake
-- Up-to-date: /usr/local/lib/python3/dist-packages/lib/pkgconfig/SoapySDR.pc
-- Up-to-date: /usr/local/lib/python3/dist-packages/bin/SoapySDRUtil
-- Up-to-date: /usr/local/lib/python3/dist-packages/share/man/man1/SoapySDRUtil.1
-- Up-to-date: /usr/local/lib/python3/dist-packages/lib/python2.7/dist-packages/_SoapySDR.so
-- Up-to-date: /usr/local/lib/python3/dist-packages/lib/python2.7/dist-packages/SoapySDR.py
-- Up-to-date: /usr/local/lib/python3/dist-packages/lib/python3.6/dist-packages/_SoapySDR.so
-- Up-to-date: /usr/local/lib/python3/dist-packages/lib/python3.6/dist-packages/SoapySDR.py

hakim@hakim-desktop:~/SoapySDR$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-g351896d3
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8 (missing)
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

We would like to use Python3 instead of python 3.6 or python 2.7 to match with our Lora module that we've got. 

Is it possible to put the SoapySDR files at this location ?

Thanks

guruofquality, 2021-01-28:
If it helps, SoapySDR/python is a stand alone cmake project that can build bindings for any python version.

Once SoapySDR library is installed:

```
mkdir build
cd build
cmake path/to/SoapySDR/python -DPYTHON_EXECUTABLE=path/to/your/python_executable
make
sudo make install
```

MartinPolytech, 2021-01-28:
Thanks for this quick answer ! 

We've tried what you said, this are the lines we wrote : 

```
hakim@hakim-desktop:~$ cd SoapySDR/
hakim@hakim-desktop:~/SoapySDR$ mkdir build
hakim@hakim-desktop:~/SoapySDR$ cd build
hakim@hakim-desktop:~/SoapySDR/build$ cmake /home/hakim/SoapySDR/python -DPYTHON_EXECUTABLE=/usr/local/lib/python3
-- The CXX compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Found SWIG: /usr/bin/swig3.0 (found version "3.0.12") 
-- SWIG_FOUND: TRUE - 3.0.12
-- Found PythonInterp: /usr/local/lib/python3 (found version "1.4") 
-- PYTHONINTERP_FOUND: TRUE - 1.4
-- PYTHON_EXECUTABLE: /usr/local/lib/python3
-- PYTHON_INSTALL_DIR: ${prefix}/
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython3.6m.so (found version "3.6.9") 
-- PYTHONLIBS_FOUND: TRUE - 3.6.9
-- PYTHON_INCLUDE_DIRS: /usr/include/python3.6m
-- PYTHON_LIBRARIES: /usr/lib/aarch64-linux-gnu/libpython3.6m.so
CMake Warning at CMakeLists.txt:83 (message):
  Python interp and library version mismatch

-- Performing Test SIZE_T_IS_UNSIGNED_INT
-- Performing Test SIZE_T_IS_UNSIGNED_INT - Failed
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hakim/SoapySDR/build
**hakim@hakim-desktop:~/SoapySDR/build$ make -j3
hakim@hakim-desktop:~/SoapySDR/build$ man make**
No manual entry for make
See 'man 7 undocumented' for help when manual pages are not available.
hakim@hakim-desktop:~/SoapySDR/build$ 
```

As you see we couldn't do the "make" command. 

We've tried to avoid the folder creation and this how we proceeded : 

```
hakim@hakim-desktop:~/SoapySDR$ cmake /home/hakim/SoapySDR/python -DPYTHON_EXECUTABLE=/usr/local/lib/python3
CMake Error: The source "/home/hakim/SoapySDR/python/CMakeLists.txt" does not match the source "/home/hakim/SoapySDR/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
hakim@hakim-desktop:~/SoapySDR$ cmake /home/hakim/SoapySDR/python/ -DPYTHON_EXECUTABLE=/usr/local/lib/python3
CMake Error: The source "/home/hakim/SoapySDR/python/CMakeLists.txt" does not match the source "/home/hakim/SoapySDR/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
hakim@hakim-desktop:~/SoapySDR$ cmake /home/hakim/SoapySDR -DPYTHON_EXECUTABLE=/usr/local/lib/python3
-- Build type not specified: defaulting to release.
-- 
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
-- SWIG_FOUND: TRUE - 3.0.12
-- Found PythonInterp: /usr/local/lib/python3 (found version "1.4") 
-- PYTHONINTERP_FOUND: TRUE - 1.4
-- PYTHON_EXECUTABLE: /usr/local/lib/python3
-- PYTHON_INSTALL_DIR: ${prefix}/lib/python2.7/dist-packages
-- PYTHONLIBS_FOUND: TRUE - 1.4
-- PYTHON_INCLUDE_DIRS: /usr/include/python2.7;/usr/include/aarch64-linux-gnu/python2.7
-- PYTHON_LIBRARIES: -L/usr/lib/python2.7/config-aarch64-linux-gnu -L/usr/lib -lpython2.7 -lpthread -ldl  -lutil -lm  -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions
-- CMAKE_SWIG_FLAGS=-c++;-threads;-I/home/hakim/SoapySDR/include
-- 
-- #############################################
-- ## Begin configuration for Python3 support...
[... 75 lines cut ...]
-- Up-to-date: /usr/local/include/SoapySDR/Version.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Formats.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Modules.h
-- Up-to-date: /usr/local/include/SoapySDR/Version.h
-- Up-to-date: /usr/local/include/SoapySDR/Device.h
-- Up-to-date: /usr/local/include/SoapySDR/Time.h
-- Up-to-date: /usr/local/include/SoapySDR/Logger.h
-- Up-to-date: /usr/local/include/SoapySDR/Config.h
-- Up-to-date: /usr/local/lib/libSoapySDR.so.0.8.0
-- Up-to-date: /usr/local/lib/libSoapySDR.so.0.8
-- Up-to-date: /usr/local/lib/libSoapySDR.so
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRExport.cmake
-- Installing: /usr/local/share/cmake/SoapySDR/SoapySDRExport-release.cmake
-- Up-to-date: /usr/local/lib/pkgconfig/SoapySDR.pc
-- Up-to-date: /usr/local/bin/SoapySDRUtil
-- Up-to-date: /usr/local/share/man/man1/SoapySDRUtil.1
-- Up-to-date: /usr/local/lib/python2.7/dist-packages/_SoapySDR.so
-- Up-to-date: /usr/local/lib/python2.7/dist-packages/SoapySDR.py
-- Up-to-date: /usr/local/lib/python3.6/dist-packages/_SoapySDR.so
-- Up-to-date: /usr/local/lib/python3.6/dist-packages/SoapySDR.py

```
It looks like SoapySDR is not in the Python3 folder as we wanted. Do you have any ideas on how to fix this ? :)

Thanks again ! Have a good evening !

guruofquality, 2021-01-28:
Just FYI, the build is looking for an executable called `${PYTHON_EXECUTABLE}-config`. This executable usually comes with python installations and tells the build where to find matching development files. If not, the build falls back to using https://cmake.org/cmake/help/latest/module/FindPythonLibs.html Therefore you have to manually tell the build where to find development files by specifying `-DPYTHON_INCLUDE_DIR=` and `-DPYTHON_LIBRARY=`

MartinPolytech, 2021-01-28:
We don't really see what you mean, do you mean the PYTHONPATH and the LD_LIBRARY_PATH as we did for the installation of GNU Radio : https://wiki.gnuradio.org/index.php/ModuleNotFoundError#B._Finding_the_Python_library ?

We apologize for taking so much time to understand but we are just beginners ^^

guruofquality, 2021-01-29:
> PYTHONPATH and the LD_LIBRARY_PATH

Those are runtime environment variables that tell python where to search, and tell the shared library loader where to search respectively. But ignore runtime for the moment, you have to first get things to correctly compile. 

Gnuradio is also a cmake project and also uses FindPythonLibs.cmake and so the cmake command to configure the project should also take `-DPYTHON_INCLUDE_DIR= and -DPYTHON_LIBRARY=`. So if you managed to get gnuradio to build against your custom python environment, I would take the same approach for SoapySDR.

Take some time, find where your development libraries and headers for your custom python environment are located. Try to pass them in using the variables mentioned, add some prints to cmake if you need to debug.

MartinPolytech, 2021-01-31:
Hello :) 

We tried doing what you said however we still have some problems. We managed to configure  the -DPYTHON3_INSTALL_DIR to python3 as  we wanted.  We also tried to change the -DPYTHON3_INCLUDE_DIRS= and the -DPYTHON3_LIBRARIES= but it didn't work (as you can see below) and when we try to execute the project, we still have the same issues.

hakim@hakim-desktop:~/SoapySDR/build$ cmake /home/hakim/SoapySDR/python3 -DPYTHON3_INSTALL_DIR=lib/python3/dist-packages -DPYTHON3_LIBRARIES=/usr/local/lib/python3 -DPYTHON3_INCLUDE_DIRS=/usr/include/python3
-- Build type not specified: defaulting to release.
-- SWIG_FOUND: TRUE - 3.0.12
-- Could NOT find Python3InterpDbg (missing: PYTHON3_DBG_EXECUTABLE) 
-- PYTHON3INTERP_FOUND: TRUE
-- PYTHON3_EXECUTABLE: /usr/bin/python3
-- PYTHON3_INSTALL_DIR: ${prefix}/lib/python3/dist-packages
-- PYTHON3LIBS_FOUND: TRUE
-- PYTHON3_INCLUDE_DIRS: /usr/include/python3.6m;/usr/include/python3.6m
-- PYTHON3_LIBRARIES: -L/usr/lib/python3.6/config-3.6m-aarch64-linux-gnu -L/usr/lib -lpython3.6m -lpthread -ldl  -lutil -lm  -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions
CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.10)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/hakim/SoapySDR/build
hakim@hakim-desktop:~/SoapySDR/build$ make -j3
hakim@hakim-desktop:~/SoapySDR/build$ 

```
We couldn't find the librairies of python3 in the /usr/include/ we only found a folder python3.6m so maybe that's what's causing the problem. We found the python librairies for the -DPYTHON_LIBRARY but we didn't manage to change the path. 

Do you have any idea how we could fix this ? We apologize again for taking so much of your time.

Have a nice evening :)

guruofquality, 2021-02-01:
> -DPYTHON3_LIBRARIES=/usr/local/lib/python3 -DPYTHON3_INCLUDE_DIRS=/usr/include/python3

These are output variables that are set by FindPythonLibs.cmake, I said to use `-DPYTHON_INCLUDE_DIR= and -DPYTHON_LIBRARY=`

From the docs: https://cmake.org/cmake/help/latest/module/FindPythonLibs.html

```
If you’d like to specify the installation of Python to use, you should modify the following cache variables:

PYTHON_LIBRARY             - path to the python library
PYTHON_INCLUDE_DIR         - path to where Python.h is found

```

Just as an example, here is a project that builds SoapySDR and GNURadio in windows, You can see the variables passed to cmake for python

* https://github.com/pothosware/PothosSDR/blob/master/BuildSoapySDR.cmake#L67
* https://github.com/pothosware/PothosSDR/blob/master/BuildGNURadio.cmake#L99

MartinPolytech, 2021-02-03:
Hello,

Thank you for your help. We managed to change the DPYTHON_INCLUDE_DIR= and -DPYTHON_LIBRARY= in the cmake. However, we still have an error when whe try to use the blocks in gnuradio :

```
Executing: /usr/bin/python3 -u /home/hakim/gr-lora/apps/lora_receive_realtime.py

Traceback (most recent call last):
  File "/home/hakim/gr-lora/apps/lora_receive_realtime.py", line 316, in <module>
    main()
  File "/home/hakim/gr-lora/apps/lora_receive_realtime.py", line 292, in main
    tb = top_block_cls()
  File "/home/hakim/gr-lora/apps/lora_receive_realtime.py", line 103, in __init__
    tune_args, settings, samp_rate, "fc32")
  File "/usr/local/lib/python3/dist-packages/soapy/soapy_swig.py", line 166, in make
    return _soapy_swig.source_make(nchan, device, dev_args, stream_args, tune_args, other_settings, sampling_rate, type)
RuntimeError: SoapySDR::Device::make() no match

```
At first, we thought this error was because we didn't  have the drivers for the hackrf. But when we call SoapySDRUtil --find we can see that they are installed correctly :

```
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 88869dc3352821b
  part_id = a000cb3c006a4767
  serial = 0000000000000000088869dc3352821b
  version = 2017.02.1

```
Do you have any insights on how we could fix this ?
Thanks again for your help. Have a great day :)

guruofquality, 2021-02-03:
You either passed some arguments to make to exclude hackrf, or you have multiple SoapySDR libraries installed; check out https://github.com/pothosware/SoapySDR/wiki/ConfigGuide#avoid-simultaneous-installs

closed 2021-06-06 by guruofquality