pothosware/SoapySDR #273: install TARGETS given no LIBRARY DESTINATION for module target

fransvanberckel, 2020-09-14:
Getting with Python3 support ...

-- PYTHON3_LIBRARIES: -L/usr/lib/python3.7/config-3.7m-arm-linux-gnueabihf -L/usr/lib -lpython3.7m -lcrypt -lpthread -ldl  -lutil -lm  -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions
-- CMAKE_SWIG_FLAGS=-c++;-threads;-I/root/workspace/SoapySDR/include;-DSIZE_T_IS_UNSIGNED_INT
CMake Error at python3/CMakeLists.txt:90 (install):
  install TARGETS given no LIBRARY DESTINATION for module target
  "_SoapySDR3".

CMake Error at python3/CMakeLists.txt:95 (install):
  install FILES given no DESTINATION!

Solved by ...

diff --git a/python3/CMakeLists.txt b/python3/CMakeLists.txt
index 6bace53..9cf7adb 100644
--- a/python3/CMakeLists.txt
+++ b/python3/CMakeLists.txt
@@ -90,6 +90,7 @@ set_target_properties(${SWIG_MODULE_SoapySDR3_REAL_NAME} PROPERTIES OUTPUT_NAME
 install(
     TARGETS ${SWIG_MODULE_SoapySDR3_REAL_NAME}
     DESTINATION ${PYTHON3_INSTALL_DIR}
+    LIBRARY DESTINATION ${PYTHON3_LIB_DIR}
 )
 
 install(

guruofquality, 2020-09-14:
Thats a strange change, perhaps PYTHON3_INSTALL_DIR ended up empty? Do you have the output from the cmake command?

It might be a failure of this script: https://github.com/pothosware/SoapySDR/blob/master/python/get_python_lib.py
You might want to run that with the same python interpreter and see whats erroring

fransvanberckel, 2020-09-14:
Sure, it's with the LIBRARY DESTINATION patch ...
```
root@raspberrypi:~/workspace/SoapySDR# mkdir build
root@raspberrypi:~/workspace/SoapySDR# cd build
root@raspberrypi:~/workspace/SoapySDR/build# cmake ..
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components:  doxygen dot 
-- 
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
-- Found SWIG: /usr/bin/swig3.0 (found version "3.0.12") 
-- SWIG_FOUND: TRUE - 3.0.12
-- Found PythonInterp: /usr/bin/python (found version "2.7.16") 
-- PYTHONINTERP_FOUND: TRUE - 2.7.16
-- PYTHON_EXECUTABLE: /usr/bin/python
-- PYTHON_INSTALL_DIR: ${prefix}/lib/python2.7/dist-packages
-- PYTHONLIBS_FOUND: TRUE - 2.7.16
-- PYTHON_INCLUDE_DIRS: /usr/include/python2.7;/usr/include/arm-linux-gnueabihf/python2.7
-- PYTHON_LIBRARIES: -L/usr/lib/python2.7/config-arm-linux-gnueabihf -L/usr/lib -lpython2.7 -lpthread -ldl  -lutil -lm  -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions
-- Performing Test SIZE_T_IS_UNSIGNED_INT
-- Performing Test SIZE_T_IS_UNSIGNED_INT - Success
-- CMAKE_SWIG_FLAGS=-c++;-threads;-I/root/workspace/SoapySDR/include;-DSIZE_T_IS_UNSIGNED_INT
CMake Warning (dev) at /usr/share/cmake-3.13/Modules/UseSWIG.cmake:564 (message):
  Policy CMP0078 is not set.  Run "cmake --help-policy CMP0078" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  python/CMakeLists.txt:155 (SWIG_ADD_LIBRARY)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- 
-- #############################################
-- ## Begin configuration for Python3 support...
-- #############################################
-- Enabling optional Python3 bindings if possible...
-- SWIG_FOUND: TRUE - 3.0.12
-- Found Python3Interp: /usr/bin/python3  
-- Found Python3InterpDbg: /usr/bin/python3-dbg  
-- PYTHON3INTERP_FOUND: TRUE
-- PYTHON3_EXECUTABLE: /usr/bin/python3
-- PYTHON3_INSTALL_DIR: ${prefix}/lib/python3.7/dist-packages
-- Found Python3Libs: -L/usr/lib/python3.7/config-3.7m-arm-linux-gnueabihf -L/usr/lib -lpython3.7m -lcrypt -lpthread -ldl  -lutil -lm  -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions  
-- PYTHON3LIBS_FOUND: TRUE
-- PYTHON3_INCLUDE_DIRS: /usr/include/python3.7m;/usr/include/python3.7m
-- PYTHON3_LIBRARIES: -L/usr/lib/python3.7/config-3.7m-arm-linux-gnueabihf -L/usr/lib -lpython3.7m -lcrypt -lpthread -ldl  -lutil -lm  -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions
-- CMAKE_SWIG_FLAGS=-c++;-threads;-I/root/workspace/SoapySDR/include;-DSIZE_T_IS_UNSIGNED_INT
CMake Warning (dev) at /usr/share/cmake-3.13/Modules/UseSWIG.cmake:564 (message):
  Policy CMP0078 is not set.  Run "cmake --help-policy CMP0078" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  python3/CMakeLists.txt:75 (SWIG_ADD_LIBRARY)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- 
-- ######################################################
-- ## SoapySDR enabled features
-- ######################################################
-- 
 * Library, runtime library v0.8.0-g580b94f3
 * Apps, command line applications
 * Tests, library unit tests
 * Docs, doxygen documentation
 * Python, python bindings v2.7.16
 * Python3, python3 bindings

-- ######################################################
-- ## SoapySDR disabled features
-- ######################################################
-- 

-- SoapySDR version: v0.8.0-g580b94f3
-- ABI/so version: v0.8
-- Install prefix: /usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /root/workspace/SoapySDR/build
```

closed 2022-11-24 by ncorgan