pothosware/SoapySDR #284: XTRX Support

RFNOD, 2020-12-02:
Hi All,

Is anyone gotten the XTRX from Fairwaves to work with Soapy? If not, I can try and figure it out and build a driver for Soapy using this guide: https://github.com/pothosware/SoapySDR/wiki/DriverGuide. I don't want to reinvent the wheel and wanted to ask first. Thanks!!

guruofquality, 2020-12-02:
Maybe it needs a link on the main panel, but give this https://github.com/xtrx-sdr/libxtrx/tree/master/soapy a try

RFNOD, 2020-12-02:
Will do. Thanks!!

hewittc, 2020-12-02:
I was only able to get their module to work after making a quick [patch](https://github.com/xtrx-sdr/libxtrx/issues/27). Without the patch, `SoapyXTRX::activateStream` wouldn't allow opening an RX stream. I can't determine if the developers are monitoring that repository or plan to provide any additional support or development for that module, though.

rfwill202, 2020-12-02:
Can you submit a PR? I can reach out to a maintainer of that repo. And they should be able to merge it.

RFNOD, 2020-12-02:
@hewittc Can you submit a PR? I can reach out to a maintainer of that repo. And they should be able to merge it. Thanks!!

RFNOD, 2020-12-02:
https://github.com/xtrx-sdr/libxtrx/pull/30 was merged.. Thanks @hewittc !!!!

RFNOD, 2020-12-02:
So, how do we add it to the SoapySDR supported devices list? Let me know what needs to be done and I can do it.

guruofquality, 2020-12-02:
> So, how do we add the XTRX to the SoapySDR supported devices list/sidebar? Let me know what needs to be done and I can do it.

I edited the wiki

closed 2020-12-02 by guruofquality