pothosware/SoapySDR #285: setFrequency variable ppm

G8BQR, 2020-12-02:
Dear Developers,

I have come across an issue which seems to be due to setFrequency variable operation.
I am trying to adjust an RSP-1a SDR receiver to within 10 Hz at 144 MHz.
When I attempt to adjust the ppm variable in OpenWebRX, which apparently links to the setFrequency variable in the SoapySDR layer, used in OWRX, I am unable to correct the RX frequency at all.
This also seems to happen with HackSDR, someone else has said.
Looking here, there is a closed issue which seems somewhat relevant but seems to have stuck on the unit of adjustment.
In the case of my RSP-1a it is ppm used in the Windows programme, which is the "native" environment.
ppm is also used in software for the rtlSDR I have, and in the HackSDR (so it seems).

The facility to adjust the RX frequency is essential for precise frequency adjustment, as crystals drift slowly with age, even if they are correct to 10 Hz in the first place :-)

Sorry If I have missed something relevant here.
I would like to encourage a high priority for the implementation of frequency correction ability in SoapySDR - if not already being addressed, otherwise SoapySDR seems to be doing what I want very well :-)

Regards,  Ed  G8BQR

zuckschwerdt, 2020-12-03:
The SDRplay Soapy driver has support for the PPM adjustment using `mir_sdr_SetPpm()` https://github.com/pothosware/SoapySDRPlay/blob/master/Settings.cpp#L487 (`setFrequency(…"CORR"…)`)

The recommended newer API (`setFrequencyCorrection()` is also available via the default implementation https://github.com/pothosware/SoapySDR/blob/master/lib/Device.cpp#L248

Maybe you can add a debug output or try setting/reading the value with other programs to see if it works?

G8BQR, 2020-12-03:
Many thanks for your response Christian,

As you probably noticed, I'm a beginner at the sort of things :-/
I know the final thing works in SDRuno, so it must be available at some point, but not working in the final OpenwebRX - I should learn a lot tracking down where it gets lost :-)
Looks like I'll be trying to put the start bits together in Ubuntu and see what happens.
And trying to see what exists in the Raspberry Pi setup. 
When all else fails I'll ask OWRX guy how it was done ;-)
Regards, Ed G8BQR

guruofquality, 2021-01-04:
Closing. Recommending using setFrequencyCorrection() or implementing it when possible.

closed 2021-01-04 by guruofquality