pothosware/SoapySDR #286: Basic Example from PythonSupport page fails with RuntimeError: setFrequency failed

RDPowerz, 2020-12-12:
So, I copied/pasted the Basic Example from https://github.com/pothosware/SoapySDR/wiki/PythonSupport into a file and ran it, and it failed.  Here is the output:

    Found Rafael Micro R820T tuner
    {default_input=True, default_output=True, device_id=0, driver=audio, label=hw:HDA Intel PCH,0}
    {default_input=False, default_output=False, device_id=13, driver=audio, label=default}
    {driver=remote, label=AirSpy HF+ [3652eb5dacde39f8], remote=tcp://192.168.1.22:55132, remote:driver=airspyhf, serial=3652eb5dacde39f8}
    {driver=rtlsdr, label=Generic RTL2832U OEM :: 00000001, manufacturer=Realtek, product=RTL2838UHIDIR, serial=00000001, tuner=Rafael Micro R820T}
    [INFO] Opening Generic RTL2832U OEM :: 00000001...
    Found Rafael Micro R820T tuner
    ('RX',)
    ('TUNER',)
    2.3999e+07, 1.764e+09
    Exact sample rate is: 1000000.026491 Hz
    [R82XX] PLL not locked!
    Traceback (most recent call last):
      File "/tmp/x.py", line 22, in <module>
        sdr.setFrequency(SOAPY_SDR_RX, 0, 912.3e6)
      File "/usr/local/lib/python2.7/dist-packages/SoapySDR.py", line 1866, in setFrequency
        return _SoapySDR.Device_setFrequency(self, *args)
    RuntimeError: setFrequency failed

I think 'PLL not locked!' is a clue.  I vaguely recall some issue with rtlsdr that only certain sample rates can be set and frequency values could be tuned, but it seems the ones in the example are not working.  My radio is a 'rtl-sdr.com v3' device purchased within the last year or so. so very mainstream, I would think.

Note I have been working in this code base for a week or two and have had success with using 'Softrock' radio with gqrx, see the Issues in SoapyAudio for some changes I suggested there.

Here is my environment.  SoapySDR and the modules are all recent code I built from the git repos.  Operating system is Ubuntu 18.04.

    $ SoapySDRUtil --find
    ######################################################
    ##     Soapy SDR -- the SDR abstraction library     ##
    ######################################################
    
    Found Rafael Micro R820T tuner
    Found device 0
      default_input = True
      default_output = True
      device_id = 0
      driver = audio
      label = hw:HDA Intel PCH,0
    
    Found device 1
      default_input = False
      default_output = False
      device_id = 13
      driver = audio
      label = default
    
    Found device 2
      driver = remote
      label = AirSpy HF+ [3652eb5dacde39f8]
      remote = tcp://192.168.1.22:55132
      remote:driver = airspyhf
      serial = 3652eb5dacde39f8
    
    Found device 3
      driver = rtlsdr
      label = Generic RTL2832U OEM :: 00000001
      manufacturer = Realtek
      product = RTL2838UHIDIR
      serial = 00000001
      tuner = Rafael Micro R820T
    
    $ SoapySDRUtil --info
    ######################################################
    ##     Soapy SDR -- the SDR abstraction library     ##
    ######################################################
    
    Lib Version: v0.8.0-g926c86d9
    API Version: v0.8.0
    ABI Version: v0.8
    Install root: /usr/local
    Search path:  /usr/local/lib/SoapySDR/modules0.8
    Module found: /usr/local/lib/SoapySDR/modules0.8/libHackRFSupport.so   (0.3.3-7d53087)
    Module found: /usr/local/lib/SoapySDR/modules0.8/libairspyhfSupport.so (0.1.1-b6cfbf5)
    Module found: /usr/local/lib/SoapySDR/modules0.8/libaudioSupport.so    (0.1.1-9aa2b5c)
    Module found: /usr/local/lib/SoapySDR/modules0.8/libremoteSupport.so   (0.6.0-c09b2f1)
    Module found: /usr/local/lib/SoapySDR/modules0.8/librtlsdrSupport.so   (0.3.1-52cb5c0)
    Available factories... airspyhf, audio, hackrf, remote, rtlsdr
    Available converters...
     -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
     -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
     -  CS32 -> [CS32]
     -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
     -  CU16 -> [CF32, CS16, CS8]
     -   CU8 -> [CF32, CS16, CS8]
     -   F32 -> [F32, S16, S8, U16, U8]
     -   S16 -> [F32, S16, S8, U16, U8]
     -   S32 -> [S32]
     -    S8 -> [F32, S16, S8, U16, U8]
     -   U16 -> [F32, S16, S8]
     -    U8 -> [F32, S16, S8]

zuckschwerdt, 2020-12-12:
1M is fine as sample rate for the RTL-SDR. The SoapyRTLSDR version 52cb5c0 has [changes to the error handling](https://github.com/pothosware/SoapyRTLSDR/commit/52cb5c0ab595adcb50262baacc6eaf9e83ae4793#diff-d3c542b2c3cb1d01a894f311212c4cf20d0fe52be1ed128b5ff2c475507e834fR349), which is what you see here.
I'll take a look what went wrong.

RDPowerz, 2020-12-12:
> I'll take a look what went wrong.
Thanks for the quick response!  I will await your findings.

zuckschwerdt, 2020-12-12:
It's not an error after all. The message is from setting the frequency correction (PPM), rtlsdr_set_freq_correction results in -2 -- which is not an error code.. https://github.com/osmocom/rtl-sdr/blob/master/src/librtlsdr.c#L927
Hang on, I'll fix this now.

zuckschwerdt, 2020-12-12:
Sorry for the trouble, that return code was really unexpected, shouldn't have missed that though.

closed 2020-12-12 by zuckschwerdt

RDPowerz, 2020-12-12:
Not a problem!  Thank you for fixing it so quickly!

I was able to pull the changes from git, build the new version of SoapyRTLSDR, install it, run the example, and see that it is fixed.
The only thing that does not seem right is "SoapySDRUtil --info" still reports the old commit hash in the version:
>  Module found: /usr/local/lib/SoapySDR/modules0.8/librtlsdrSupport.so   (0.3.1-24b27fa)
It should end with 'ed22', no?
Yet the fix is working so I know the changed library has been installed.
Do I need to rebuild the main SoapySDR directory for it to pick up the new version string from the library?

Thanks again,RDP

zuckschwerdt, 2020-12-13:
You need to wipe the `build` directory for SoapyRTLSDR, or at least the `build/Version.cpp` file, then rebuild and install `librtlsdrSupport.so` to get the correct version tag shown.

RDPowerz, 2020-12-13:
Thank you, I see my mistake now.
By the way I uploaded an extended Python SoapySDR example program at https://github.com/RDPowerz/SoapyPythonExamples along with its output for three radios I have access to.  It is free for anyone to use as they see fit.
Regards,RDP

guruofquality, 2020-12-13:
> You need to wipe the build directory for SoapyRTLSDR, or at least the build/Version.cpp file, then rebuild and install librtlsdrSupport.so to get the correct version tag shown.

Just FYI The cmake build extracts the info from git and puts it into the file. Simply pulling the repo wont cause cmake to re-run (unless there were CMakeLists.txt changes). You can touch CMakeCache.txt or rerun cmake to get the same effect.

RDPowerz, 2020-12-28:
Hi, Josh.
I raised an issue ( https://github.com/pothosware/SoapyAudio/issues/16 ) against SoapyAudio that includes a suggested patch to address the issue.  I'm wondering if that issue is being considered, and am wondering if a pull request would be helpful.
Regards,RDP

guruofquality, 2020-12-28:
The hamlib stuff was sort of tacked on, and it seems to be pretty specific to certain devices. Im not very familiar with it, but yes, a PR would be helpful. I'm happy to have it.

RDPowerz, 2020-12-29:
Thanks, Josh.  I did try to code defensively i.e. tried hard to not break any existing functionality.  
I will send a PR some time in the next few days depending on when I get a chance to do it.
Regards,RDP

RDPowerz, 2021-03-22:
Three months late, but I did just send a pull request last night...