pothosware/SoapySDR #287: SDRPlay with Soapysdr

Konradrundfunk, 2021-01-02:
Hi, I own a SDRPlay RSP1A and I have installed the corresponding Plugin onto my Linux Mint 20 system. 
When I run  `SoapySDRUtil --probe="driver=sdrplay"` everything is listed as intended. But when I run your example code for the api only my rtl sdr is listed. Is that a bug within the Plugin ?

Konradrundfunk, 2021-01-02:
That's my slightly modified code:
`#include <cstdio>	//stdandard output
#include <cstdlib>

#include <SoapySDR/Device.hpp>
#include <SoapySDR/Types.hpp>
#include <SoapySDR/Formats.hpp>

#include <string>	// std::string
#include <vector>	// std::vector<...>
#include <map>		// std::map< ... , ... >

#include <iostream>

int main()
{

	// 0. enumerate devices (list all devices' information)
	SoapySDR::KwargsList results = SoapySDR::Device::enumerate();
	SoapySDR::Kwargs::iterator it;

	for( int i = 0; i < results.size(); ++i)
	{
		printf("Found device #%d: ", i);
		for( it = results[i].begin(); it != results[i].end(); ++it)
		{
			printf("%s = %s\n", it->first.c_str(), it->second.c_str());
		}
		printf("\n");
	}
	printf("Done\n");

	return EXIT_SUCCESS;
}
`

guruofquality, 2021-01-02:
typically when the API and command line tools show different results you probably have multiple installs and or version of SoapySDR

Konradrundfunk, 2021-01-02:
Ok how can I check or resolve ?

fventuri, 2021-01-02:
@Konradrundfunk - please run these commands and post here their output:

```
SoapySDRUtil --find
type -a SoapySDRUtil
ldd <full path of SoapySDRUtil from the command above>
ldd <your compiled executable that you used for your test above>
```

Thanks,
Franco

Konradrundfunk, 2021-01-02:
Soapysdr find
`######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  driver = sdrplay
  label = SDRplay Dev0 RSP1A 200601BE97

konrad@konrad-ThinkPad-L380-Y`
The directory
`SoapySDRUtil ist /usr/local/bin/SoapySDRUtil`
The ldd from the Soapysdr dir
`ldd /usr/local/bin/SoapySDRUtil
	linux-vdso.so.1 (0x00007ffe3f54a000)
	libSoapySDR.so.0.8 => /usr/local/lib/libSoapySDR.so.0.8 (0x00007fdcca968000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fdcca787000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fdcca76c000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fdcca749000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fdcca557000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fdcca551000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fdcca400000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fdccaa26000)
`
Here the ldd from the Example compiled
`ldd src/Example 
	linux-vdso.so.1 (0x00007ffecb150000)
	libSoapySDR.so.0.7 => /usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7 (0x00007f1d5941d000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f1d5923c000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f1d59221000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f1d5902f000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f1d59029000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f1d58eda000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f1d58eb5000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f1d594b1000)
`

Konradrundfunk, 2021-01-02:
Seams like i have two installs ok thanks a lot for helping!

closed 2021-01-02 by Konradrundfunk