pothosware/SoapySDR #293: Build fails on Mac because of -lintl

treaves, 2021-02-05:
Mac OS Big Sur, Python 3.9 installed with brew.

The `-lintl` is pulled in by the `FindPython3Libs.cmake` file.  I see it's a common issue, but, I'm unclear of the solution. 

```
[ 97%] Linking CXX shared module _SoapySDR.so
[ 97%] Built target _SoapySDR
1 warning generated.
[100%] Linking CXX shared module _SoapySDR.so
ld: library not found for -lintl
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

guruofquality, 2021-02-07:
Its weird because the link flags should only come from the python-config executable also install by homebrew, so python shouldnt ask for flags it doesnt have libraries as dependencies for. Anyway... Does installing the missing library fix the problem? `brew install intltool`

treaves, 2021-02-07:
Sorry, I should have said I'd already tried that.

The dylib's are there.  Not sure why this is happening.

closed 2022-11-24 by ncorgan