pothosware/SoapySDR #295: Ubuntu 20.10 make -j4 fails

Sszymajda, 2021-03-28:
Hello, i have trouble building SoapySDR for SDRAngel, im following instruction from build guide and it still fails im using ubuntu 20.10. everything seems all right but when i try do make it fails on 97% 
"[ 97%] Linking CXX shared module _SoapySDR.so
/usr/bin/ld: CMakeFiles/_SoapySDR3.dir/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx.o: in function `SWIG_Python_ErrorType(int)':" 
im still pretty basic so its could be some stupid error made by me so sorry for that xp 
anyway if anybody could help with this issue 
sincerly simon
PS im attatching log file from make log
[build.log](https://github.com/pothosware/SoapySDR/files/6217143/build.log)

guruofquality, 2021-03-28:
Either this is just some weird config issues with finding the python library or it might be this new dynamic lookup thing in the newer python libraries.

Can you post the output of cmake? Make sure it has reasonable includes and libs for python3.

Recently OSX made me add "-undefined dynamic_lookup" flag: https://github.com/pothosware/SoapySDR/blob/master/python/CMakeLists.txt#L158 Try adding this flag: `-unresolved-symbols=ignore-all`. So if thats the case, I can add it to the build..

Sszymajda, 2021-03-30:
hello, no problem i have tried and it still fails. with this flag :C 
those are two dump files one with flag one without one 
[cmake_dump.txt](https://github.com/pothosware/SoapySDR/files/6232202/cmake_dump.txt)
[cmake_dump_unresolved_symobols.txt](https://github.com/pothosware/SoapySDR/files/6232203/cmake_dump_unresolved_symobols.txt)

guruofquality, 2021-03-30:
I see the problem there is no .so file being linked against in the python3 build. Which I think python wants us to do these days

-- PYTHON_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython2.7.so
-- PYTHON3_LIBRARIES: -L/usr/lib/python3.8/config-3.8-x86_64-linux-gnu -L/usr/lib  -lcrypt -lpthread -ldl  -lutil -lm -lm

Can you try the `-unresolved-symbols=ignore-all` flag in this file actually (its the python3 directory I linked the wrong file) https://github.com/pothosware/SoapySDR/blob/master/python3/CMakeLists.txt#L158

Sszymajda, 2021-03-31:
hello, okay so what i did, i used this command 

sudo cmake -DCMAKE_INSTALL_PREFIX=/opt/install/SoapySDR --PYTHON_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython2.7.so --PYTHON3_LIBRARIES: -L/usr/lib/python3.8/config-3.8-x86_64-linux-gnu -L/usr/lib -lcrypt -lpthread -ldl -lutil -lm -lm -undefined dynamic_lookup .. 

there is attached file with cmake output with this command, and if i followed correctly i copied content from this link to build/python3/CMakeLists.txt (well i know probably its stupid mistake im still learning :) )
https://github.com/pothosware/SoapySDR/blob/master/python3/CMakeLists.txt#L158

after i copied this file there was more progress when i tried make -j4 install and it failed again 😭 
also im attaching log for this 

[cmake_dump.txt](https://github.com/pothosware/SoapySDR/files/6239765/cmake_dump.txt)
[make_j4_install_dump.txt](https://github.com/pothosware/SoapySDR/files/6239766/make_j4_install_dump.txt)

closed 2021-04-01 by guruofquality