pothosware/SoapySDR #296: Failing to load driver modules using python interface

MichelDequick, 2021-03-29:
When running following code:
```pyton
import SoapySDR
from SoapySDR import *

args = dict(driver="hackrf")
sdr = SoapySDR.Device(args)
```
I get this error:
```pyton
[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspyhfSupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspySupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/HackRFSupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/LMS7Support.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/miriSupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/osmosdrSupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/PlutoSDRSupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/rtlsdrSupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/sdrPlaySupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/uhdSupport.dll)       
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/volkConverters.dll)   
  LoadLibrary() failed: The specified module could not be found.

Traceback (most recent call last):
  File "c:\Users\michel\Desktop\test\main.py", line 42, in <module>
    sdr = SoapySDR.Device(args)
  File "C:\Program Files\PothosSDR\lib\python3.9\site-packages\SoapySDR.py", line 1838, in __new__    
    return cls.make(*args, **kwargs)
  File "C:\Program Files\PothosSDR\lib\python3.9\site-packages\SoapySDR.py", line 1454, in make       
    return _SoapySDR.Device_make(*args)
RuntimeError: SoapySDR::Device::make() no match
```
The code was run on a Dell XPS 15 9560 with Windows 10 as operating system.
A hackrf with the newest drivers was connected to the laptop.
This issue does not appear when running the same code on a desktop.
I tryed a clean install of Potohos & Python 3.9 with no succes

Thanks in advance

guruofquality, 2021-03-30:
Does the command line utility do the same thing? `C:\Program Files\PothosSDR\bin\SoapySDRUtil --make="driver=hackrf"`
Or if it fails, does it work if you `cd C:\Program Files\PothosSDR\bin` first?

Failing to load modules that have been discovered in the file system usually means some dll is not in the path. I have no idea whats happening --

As an uninformed guess, it could have something to do with python and `os.add_dll_directory`. At the top of C:\Program Files\PothosSDR\lib\python3.9\site-packages\SoapySDR.py, there should be some code to add the dll search path invoked with `os.add_dll_directory(C:\Program Files\PothosSDR\bin)`

MichelDequick, 2021-03-30:
Running `SoapySDRUtil --make="driver=hackrf"` in the right directory executes without errors.

```cmd
SoapySDRUtil --make="driver=hackrf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Make device driver=hackrf
[INFO] Opening HackRF One #0 XXXXXXXXXXXXX
  driver=HackRF
  hardware=HackRF One
  clock source=external
  part id=XXXXXXXXXXXXXXX
  serial=00000000000000000XXXXXXXXXXXXX
  version=local-d6f9d40
```

Issue is very likely to be from the python module loading. Any clue on how to debug or approach this?

guruofquality, 2021-03-30:
Is the python 3.9 from the amd64 installer? I somehow think it would fail importing if it wasnt. Just a guess. https://www.python.org/ftp/python/3.9.2/python-3.9.2-amd64.exe

You could try being very explicit about the python being used to make sure its not a different one from some other package.

For example, this should work, and not print crazy stuff
`C:\Program Files\Python\Python3.9\python.exe -c "import SoapySDR; print(SoapySDR.Device.enumerate())"`

MichelDequick, 2021-04-06:
It fixed itself, thanks for the help!
¯\_(ツ)_/¯

closed 2021-04-06 by MichelDequick