pothosware/SoapySDR #297: Failed to create Soapy object

northernbelysh, 2021-04-09:
## Discription
Hello!
I am trying to start srsenb with the LimeSDR Q-PCi, but I get the following errors from Soapy (errors are in the last two lines):

`Reading configuration file ue.conf.example...`

`WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.`

`Built in Release mode using commit 45486b6e2 on branch master.`
`Opening 1 channels in RF device=soapy with args=rxant=LNAH,txant=BAND2`
`Soapy has found device #0: driver=lime, index=1, label=LimeSDR-QPCIe [PCI-E], media=PCI-E, module=PCIEXillybus, name=LimeSDR-QPCIe,`
`Selecting Soapy device: 0`
`[INFO] Make connection: 'LimeSDR-QPCIe [PCI-E]'`
`[INFO] CLK0 fOut = 27 MHz  Multisynth Divider 33 0/1  R divider = 1 source = PLLA`
`[INFO] CLK1 fOut = 27 MHz  Multisynth Divider 33 0/1  R divider = 1 source = PLLA`
`[INFO] CLK2 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA`
`[INFO] CLK3 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA`
`[INFO] CLK4 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA`
`[INFO] CLK5 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA`
`[INFO] CLK6 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA`
`[INFO] CLK7 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA`
`[INFO] Si5351C: VCOA = 891 MHz  Feedback Divider 35 41943/65536`
`[INFO] Si5351C: VCOB = 891 MHz  Feedback Divider 35 41943/65536`
`[INFO] Device name: LimeSDR-QPCIe`
`[INFO] Reference: 30.72 MHz`
`[INFO] LMS7002M calibration values caching Disable`

`Failed to create Soapy object`
`Error initializing radio.`

## Additional information
I installed the program according to the official instructions of the developer MyRiadrf
### LimeUtil --info

LimeSuite information summary
Version information:
  Library version:      v18.10.1-gb8b4bcab
  Build timestamp:      2021-04-01
  Interface version:    v2018.10.0
  Binary interface:     18.10-1

System resources:
  Installation root:    /usr/local
  User home directory:  /home/lte
  App data directory:   /home/lte/.local/share/LimeSuite
  Config directory:     /home/lte/.limesuite
  Image search paths:
     - /home/lte/.local/share/LimeSuite/images
     - /usr/local/share/LimeSuite/images

Supported connections:
   * FT601
   * FX3
   * PCIEXillybus
   
### LimeUtil --find 
`* [LimeSDR-QPCIe, media=PCI-E, module=PCIEXillybus, index=1]`

### SoapySDRUtil --info
 Soapy SDR -- the SDR abstraction library     
Lib Version: v0.7.1-myriadrf1~ubuntu18.04
API Version: v0.7.1
ABI Version: v0.7
Install root: /usr
Search path:  /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7        (missing)
Search path:  /usr/local/lib/x86_64-linux-gnu/SoapySDR/modules0.7  (missing)
Search path:  /usr/local/lib/SoapySDR/modules0.7
Module found: /usr/local/lib/SoapySDR/modules0.7/libLMS7Support.so (18.10.1-b8b4bcab)
Available factories... lime
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

### SoapySDRUtil --find

Found device 0
  driver = lime
  index = 1
  label = LimeSDR-QPCIe [PCI-E]
  media = PCI-E
  module = PCIEXillybus
  name = LimeSDR-QPCIe

### SoapySDRUtil --probe

Probe device
[INFO] Make connection: 'LimeSDR-QPCIe [PCI-E]'
[INFO] CLK0 fOut = 27 MHz  Multisynth Divider 33 0/1  R divider = 1 source = PLLA
[INFO] CLK1 fOut = 27 MHz  Multisynth Divider 33 0/1  R divider = 1 source = PLLA
[INFO] CLK2 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA
[INFO] CLK3 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA
[INFO] CLK4 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA
[INFO] CLK5 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA
[INFO] CLK6 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA
[INFO] CLK7 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA
[INFO] Si5351C: VCOA = 891 MHz  Feedback Divider 35 41943/65536
[INFO] Si5351C: VCOB = 891 MHz  Feedback Divider 35 41943/65536
[INFO] Device name: LimeSDR-QPCIe
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M calibration values caching Disable
Error probing device: vector::_M_range_check: __n (which is 2) >= this->size() (which is 2)

guruofquality, 2021-04-09:
This is really a job for https://github.com/myriadrf/LimeSuite/issues/

If there is a bound check issue in the driver, the maintainers should know about it

northernbelysh, 2021-04-09:
> Это действительно работа для https://github.com/myriadrf/LimeSuite/issues/
> 
> Если в драйвере есть проблема связанной проверки, разработчики должны знать об этом.

Спасибо за быстрый ответ. Я им тоже задам этот вопрос. Но пока не буду здесь закрывать. Может кто встречал эту проблему.

closed 2022-11-24 by ncorgan