pothosware/SoapySDR #299: Basic C API Example doesn't compile

shredEngineer, 2021-04-20:
The [Basic C API Example](https://github.com/pothosware/SoapySDR/wiki/C_API_Example) doesn't compile under Ubuntu 20.04 (64-Bit).

```
gcc -std=c99 -o test test.c -lSoapySDR
test.c: In function ‘main’:
test.c:71:37: warning: passing argument 2 of ‘SoapySDRDevice_setupStream’ makes integer from pointer without a cast [-Wint-conversion]
   71 |     SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, SOAPY_SDR_CF32, NULL, 0, NULL);
      |                                     ^~~~~~~~~
      |                                     |
      |                                     SoapySDRStream ** {aka struct SoapySDRStream **}
In file included from test.c:1:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘int’ but argument is of type ‘SoapySDRStream **’ {aka ‘struct SoapySDRStream **’}
  295 | SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
test.c:71:48: warning: passing argument 3 of ‘SoapySDRDevice_setupStream’ makes pointer from integer without a cast [-Wint-conversion]
   71 |     SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, SOAPY_SDR_CF32, NULL, 0, NULL);
      |                                                ^~~~~~~~~~~~
      |                                                |
      |                                                int
In file included from test.c:1:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const char *’ but argument is of type ‘int’
  295 | SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
test.c:71:62: warning: passing argument 4 of ‘SoapySDRDevice_setupStream’ from incompatible pointer type [-Wincompatible-pointer-types]
   71 |     SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, SOAPY_SDR_CF32, NULL, 0, NULL);
      |                                                              ^~~~~~~~~~~~~~
      |                                                              |
      |                                                              char *
In file included from test.c:1:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const size_t *’ {aka ‘const long unsigned int *’} but argument is of type ‘char *’
  295 | SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
test.c:71:78: warning: passing argument 5 of ‘SoapySDRDevice_setupStream’ makes integer from pointer without a cast [-Wint-conversion]
   71 | evice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, SOAPY_SDR_CF32, NULL, 0, NULL);
      |                                                                 ^~~~
      |                                                                 |
      |                                                                 void *

In file included from test.c:1:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘size_t’ {aka ‘const long unsigned int’} but argument is of type ‘void *’
  295 | SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
test.c:71:5: error: too many arguments to function ‘SoapySDRDevice_setupStream’
   71 |     SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, SOAPY_SDR_CF32, NULL, 0, NULL);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from test.c:1:
/usr/local/include/SoapySDR/Device.h:295:31: note: declared here
  295 | SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
      |
```

BTW: Just a few minutes before this, I successfully built SoapySDR, SoapySDRPlay and PothosCore from scratch.

guruofquality, 2021-04-21:
There was an API change between release and the current master branch. The comments in the wiki are basically there to support both versions: 

```
//setup a stream (complex floats)
    // SoapySDRStream *rxStream = SoapySDRDevice_setupStream(sdr, SOAPY_SDR_RX, SOAPY_SDR_CF32, NULL, 0, NULL);
    // not compiled at least since 0.7.1
    // use below instead { // just for build 
    SoapySDRStream *rxStream;
    SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, SOAPY_SDR_CF32, NULL, 0, NULL);
    // }
    ```

ncorgan, 2021-04-22:
This seems like something to be added to the CI. Examples should compile with the version they build against.

guruofquality, 2021-04-23:
Well looks like there is some desire to have a tag with the recent gr-soapy work and SoapySDRPlay3. So I updated the example.

> This seems like something to be added to the CI. Examples should compile with the version they build against.

Yea it should probably live in the repo and just get linked to in the wiki

closed 2021-04-26 by guruofquality