pothosware/SoapySDR #303: "Command not found" for any rx_*

BrianLowe, 2021-04-26:
I also left this on the rx_tools page but with no answers:

I have SoapySDR successfully implemented onto Raspbian but cannot figure out how to get any rx_* command to work. I am relatively new to SDR's and couldn't seem to figure out what is going wrong.

I always get this error message in terminal:
rx_fm: command not found

Help ASAP would be greatly appreciated!

zuckschwerdt, 2021-04-26:
You didn't (properly) install rx_tools. Try https://github.com/rxseger/rx_tools#installation -- then `sudo make install`or call the programs as `./rx_fm`
Raspberry Pi OS (previously called Raspbian) doesn't feature a rx_tools packages AFAIK.

closed 2021-04-26 by zuckschwerdt

BrianLowe, 2021-04-26:
> You didn't (properly) install rx_tools. Try https://github.com/rxseger/rx_tools#installation -- then `sudo make install`or call the programs as `./rx_fm`
> Raspberry Pi OS (previously called Raspbian) doesn't feature a rx_tools packages AFAIK.

Hi,

I followed the installation instructions and SoapySDR is working fine. I run `cmake .` `make` and `sudo make install` and all succeed but I still cannot get rx_fm to work. After following those three steps it looks like this:

`pi@raspberrypi: ~/SoapySDR $ rx_fm
bash: rx_fm: command not found
pi@raspberrypi: ~/SoapySDR $ ./rx_fm
bash: ./rx_fm: No such file or directory`

zuckschwerdt, 2021-04-26:
It's not that `rx_fm` isn't working, it's just not found / not installed. Just to make sure: did you install `rx_tools`? It looks like you are in a `SoapySDR` directory, which likely is something SoapySDR and not rx_tools? (the implied first step of the rx_tools installation should be "download rx_tools from tgz and unpack or clone with git, then change to that directory.")

BrianLowe, 2021-04-26:
> It's not that `rx_fm` isn't working, it's just not found / not installed. Just to make sure: did you install `rx_tools`? It looks like you are in a `SoapySDR` directory, which likely is something [...]

Yes, I have rx_tools downloaded and in its own directory but there are no executables for any of the rx_* functions in there as well.

zuckschwerdt, 2021-04-26:
The steps are: download, unpack, **change to that new directory**, configure, build, install.
Change to the rx_tools directory, then `cmake . ; make ; sudo make install`