pothosware/SoapySDR #305: Undefined symbol: _ZN5boost6chrono12steady_clock3nowEv

izabarovsky, 2021-05-15:
Hi!  Get error when start CubicSdr, or SdrAngel, or even gnuradio-companion:

```
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
[ERROR] SoapySDR::loadModule(/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so)
  dlopen() failed: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/librfspaceSupport.so: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv
Hash collision!!! Fatal error!!
pure virtual method called
terminate called without an active exception
Aborted
```
Can anybody help me?

gnuradio version: 3.8.2.0 (Python 3.9.2)

guruofquality, 2021-05-15:
I didnt get a lot of information about what you installed, so this is all just a guess

Its possible that librfspace from the package manager is just kind of funny, so you could try to uninstall it

More likely this is just some abi library error from a bad install, I would encourage you to read:

* https://github.com/pothosware/SoapySDR/wiki/ConfigGuide#what-to-install
* https://github.com/pothosware/SoapySDR/wiki/ConfigGuide#avoid-simultaneous-installs

izabarovsky, 2021-05-15:
purge all previously installed packages, then install from debian repos:
```apt install install libsoapysdr-dev
...
apt install sdrangelove
```
Same error
May be installation from src will help?

guruofquality, 2021-05-15:
> May be installation from src will help?

Maybe, im not sure whats wrong with the package. But I would at least try to uninstall just the deb package that provides the librfspaceSupport.so module first. Maybe its just one bad lib, and everything else is fine. You can always install an individual rfspace module from source if its needed.

izabarovsky, 2021-05-16:
Fixed with installation from sources
Install gnuradio v3.10
Install gr-osmosdr gr-osmosdr 0.2.0.0 
Custom path used:
`cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/opt/gnuradio-release -DPYTHON_EXECUTABLE=/usr/bin/python ../
`
`cmake -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python3 ../ -DCMAKE_INSTALL_PREFIX=/opt/gr-osmosdr-release
`
So some magic required in .bashrc
`export PYTHONPATH=/opt/gnuradio-release/lib/python3/dist-packages:$PYTHONPATH
export LD_LIBRARY_PATH=/opt/gnuradio-release/lib:$LD_LIBRARY_PATH
export GRC_BLOCKS_PATH=/opt/gr-osmosdr-release/share/gnuradio/grc/blocks`

And link 
`sudo ln -s /opt/gnuradio-release/bin/gnuradio-companion /usr/bin/gnuradio-companion`

Now gnuradio-companion + HackRf transmitt whithout any errors. But any another soft (like CubicSdr or sdrangel) reinstall required

closed 2021-05-16 by izabarovsky