pothosware/SoapySDR #306: [low-priority] Swig compile failure

xloem, 2021-05-16:
after `cmake ..` on git master today:
```
[ 94%] Swig compile /home/user/src/SoapySDR/build/python/SoapySDR.i for python
/home/user/src/SoapySDR/build/python/SoapySDR.i:38: Error: Syntax error in input(1).
```

guruofquality, 2021-05-16:
Looks related to the %pythonbegin %{ ... %} section. Perhaps its a swig 3.0 thing. I can add a version check but im not 100% sure thats the reason. Do you have a version of swig3.0 in the package manager to try, for example ubuntu is `apt-get install swig3.0`

guruofquality, 2021-05-16:
FYI `pythonbegin` was added in `Version 2.0.11 (15 Sep 2013)` http://swig.org/Release/CHANGES

marcusmueller, 2021-06-08:
To be honest, I remember Ubuntu 14.04 already shipping swig 2.0.11, and everything after that shipped swig 3.0 or later at will. Any system that's more antiquated than that is probably not worth porting new software to.

guruofquality, 2021-06-09:
One alternative to the version check would be to add swig ifdefs around the `pythonbegin` section using SWIG_VERSION http://www.swig.org/Doc4.0/Preprocessor.html

closed 2022-11-24 by ncorgan

xloem, 2022-11-24:
it is humorous that for a year and a half three of us chose to discuss and postpone this rather than simply patching a one line fix. ncorgan figured this out.