pothosware/SoapySDR #324: Travis CI needs to move to travis-ci.com

ncorgan, 2021-09-18:
https://travis-ci.org/pothosware/SoapySDR

> Since June 15th, 2021, the building on travis-ci.org is ceased. Please use travis-ci.com from now on.

ncorgan, 2021-09-18:
Of course, unless we move over to Github Actions, which may be better

zuckschwerdt, 2021-09-18:
Did you already sketch something with Actions? https://github.com/pothosware/SoapySDR/pull/319#issuecomment-877673047

ncorgan, 2021-09-18:
> Did you already sketch something with Actions? #319 (comment)

Unfortunately, it fell by the wayside. I suppose this is new motivation.

zuckschwerdt, 2021-09-18:
I know that too well ;) Maybe our build_check_job (CMake+Ninja/UnixMakefiles) and analyzer_check_job (Clang) can be a quick starting point: https://github.com/merbanan/rtl_433/blob/master/.github/workflows/check.yml

closed 2021-09-28 by ncorgan