pothosware/SoapySDR #336: SDRPlay [ERROR] error in activateStream() - Init() failed: sdrplay_api_ServiceNotResponding

paulh002, 2021-12-20:
Hi,

I use SoaySDR in my own software with several SDR receivers and that all works fine. Only SDRPLay SoapySDR driver keep getting an error. [ERROR] error in activateStream() - Init() failed: sdrplay_api_ServiceNotResponding
I have the same error when I use SoapySDRUtil when I run it in the debugger. When I start SoapySDRUtil from the commandline it works. So it seems I am missing something. Any tips?
The environment I use is Raspberry Pi4 buster linux 32 bits. I use Visual Studio (with visualgdb) to crosscompile and debug.

paulh002, 2021-12-20:
Hello Franko,

Your right, now SoapySDRUtil works, that is when I don't have breakpoints. But unfortuanaly I have this issue in my client application. That why I tried to build and debug SoapySDRUtil.

fventuri, 2021-12-20:
@paulh002 - in gdb (which is what I use on Linux), I typically use breakpoints with commands (see here: https://sourceware.org/gdb/current/onlinedocs/gdb/Break-Commands.html#Break-Commands), where the last command before `end` is always a `cont` (continue) to avoid the problem with the heartbeat.

Franco

paulh002, 2021-12-20:
@fventuri Maybe you missed. I am developing for linux on raspberry pi. But use VisualGDB extension for VisualStudio. 
The issue I have is I get this error in my application on the raspberry pi. I looks like my application cannot find the  libsdrplay_api. Where as for other SDR receivers it works fine

SDRplay, 2021-12-20:
If you just want debug output from the API but you're running the application without stopping, add a call to sdrplay_api_DebugEnable after sdrplay_api_SelectDevice and sdrplay_api_UnlockDeviceApi (but before the call to sdrplay_api_GetDeviceParams

If you want to be able to step through your application, then you can disable the API heartbeat using a call to sdrplay_api_DisableHeartbeat() before sdrplay_api_SelectDevice

However, make sure to not leave either of these calls into any production code or you may run into issues.

Hope that helps,

Andy

paulh002, 2021-12-20:
The issue happens when calling sdrplay_api_Init in ActivateStream()
 err = sdrplay_api_Init(device.dev, &cbFns, (void *)this);
    if (err != sdrplay_api_Success)
    {
        SoapySDR_logf(SOAPY_SDR_ERROR, "error in activateStream() - Init() failed: %s", sdrplay_api_GetErrorString(err));
        return SOAPY_SDR_NOT_SUPPORTED;
    }
I just get an error failed.  I made a test application where I converted the code from SoapySDRUtil to support classes so I can use it in my code, and this works fine for Adalm Pluto, rtlsdr etc. But with sdrplay I get this error.
The question is why is it working ok in SoapySDRUtil and not in my version. I shared the code on github. https://github.com/paulh002/SoapySDRTest

SDRplay, 2021-12-20:
What parameters are you calling your app with? The same as per SoapySDRUtil?

So it runs ok before your changes, but not after?

Without any API debug output, I can't really say why the init would be failing.

Best regards,

Andy

fventuri, 2021-12-20:
@paulh002 - as Andy said, a common reason for `sdrplay_api_Init()` to fail is that one (or more) device configuration parameters are invalid (or out of range), like sample rate, gains, etc.

A good way to see why `sdrplay_api_Init()` failed is to search for messages like the one below in the Raspberry Pi log files (you can use something like `grep sdrplay_apiService /var/log/messages | grep ERROR` or something similar with the output from `journalctl`):

```
Jul 27 20:51:21 fvdesktop sdrplay_apiService[3669]: [7243]: devIdx0: sdrplay_apiService_rsp: checkTunerParamLimits: ERROR: gain gRdB out of range (1:20) = 59
```

Franco

paulh002, 2021-12-21:
Here the log, I see an error in the setGR function. If I compare this log with a correct one

Dec 21 09:26:58 raspberrypi SoapySDRTest: [1422]: sdrplay_api_Open
Dec 21 09:26:58 raspberrypi kernel: [  800.731693] usb 1-1.2.2: USB disconnect, device number 6
Dec 21 09:26:59 raspberrypi kernel: [  801.133292] usb 1-1.2.2: new high-speed USB device number 9 using xhci_hcd
Dec 21 09:26:59 raspberrypi kernel: [  801.364105] usb 1-1.2.2: New USB device found, idVendor=1df7, idProduct=3000, bcdDevice= 2.07
Dec 21 09:26:59 raspberrypi kernel: [  801.364126] usb 1-1.2.2: New USB device strings: Mfr=0, Product=0, SerialNumber=1
Dec 21 09:26:59 raspberrypi kernel: [  801.364145] usb 1-1.2.2: SerialNumber: 2006109397
Dec 21 09:26:59 raspberrypi mtp-probe: checking bus 1, device 9: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.2"
Dec 21 09:26:59 raspberrypi mtp-probe: bus: 1, device: 9 was not an MTP device
Dec 21 09:26:59 raspberrypi mtp-probe: checking bus 1, device 9: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.2"
Dec 21 09:26:59 raspberrypi mtp-probe: bus: 1, device: 9 was not an MTP device
Dec 21 09:26:59 raspberrypi sdrplay_apiService: [1471]: sdrplay_apiService_device: mCmdHandlerThread: Entry
Dec 21 09:26:59 raspberrypi sdrplay_apiService: [1472]: sdrplay_apiService_rsp: streamThread: Entry
Dec 21 09:26:59 raspberrypi SoapySDRTest: [1473]: sdrplay_api_device: strmHandlerThread: Entry
Dec 21 09:26:59 raspberrypi SoapySDRTest: [1474]: sdrplay_api_device: eventHandlerThread: Entry
Dec 21 09:26:59 raspberrypi sdrplay_apiService: [1471]: sdrplay_apiService_rsp_log_enable 3
Dec 21 09:26:59 raspberrypi SoapySDRTest: [1475]: sdrplay_api_device: heartBeatThread: Entry
Dec 21 09:26:59 raspberrypi sdrplay_apiService: [507]: sdrplay_apiService_log_enable: 0
Dec 21 09:26:59 raspberrypi sdrplay_apiService: [1471]: sdrplay_apiService_rsp_log_enable 0
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [507]: sdrplay_apiService_log_enable: 1
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: sdrplay_apiService_rsp_log_enable 1
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: initNumActivePackets: NumActivePackets = 4
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: initDownConverterState: Tuner1: DownConvert: Enable=1 DecM=3 OutScale=3 (fs=6.000000 bw=600 if=1620)
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: initDownConverterState: Tuner1: DownConvert: Ionly = 0 Lif = 1 (1580000 1620000) adsbEn = 0 adsbMode = 0
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: initNumActivePackets: NumActivePackets = 12
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: bridge_SetRegister: bridge_SetRegister(0x08, 0x00ee80)
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: usb_Init: usb_Init()
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: usb_SetInterface: usb_SetInterface(3)
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: bridge_Init: bridge_Init: fnaddr = 5
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: bridge_SetRegister: bridge_SetRegister(0x05, 0x00000c)
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: bridge_SetRegister: bridge_SetRegister(0x09, 0x00000e)
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: bridge_SetRegister: bridge_SetRegister(0x00, 0x000200)
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: bridge_Init: bridge_Init: adjusting squelch trim 0x1, rx gating enable 1, tx_trim 0, reg2 = 0x4801
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: bridge_SetRegister: bridge_SetRegister(0x02, 0x004801)
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: rsp1aBiasTControl: biasTEnable = 0
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: rsp1aDabNotchEnable: rfDabNotchEnable = 0
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: rsp1aBroadcastNotchEnable: rfNotchEnable = 0
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: rsp1aRfTest: rfTestEn = 0, rfTestSrcStability = 0
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: bridge_SetRegister: bridge_SetRegister(0x09, 0x28bb85)
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: bridge_SetRegister: bridge_SetRegister(0x07, 0x000494)
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: initFsFreqDependentHw: initFsFreqDependentHw: Register7 = 0x000494
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: setTunerDc: Tuner1: DCCAL: mode->4 speedup->0 (applied at next gain update)
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: setTunerDc: Tuner1: DCTRACK_TIM->63 DCREFRESHRATE_TIM->2048
Dec 21 09:27:04 raspberrypi sdrplay_apiService: [1471]: devIdx0: sdrplay_apiService_rsp: bridge_SetRegister: bridge_SetRegister(0x09, 0x2003f6)
[... cut ...]
Dec 21 09:27:14 raspberrypi sdrplay_apiService: [519]: devIdx0: sdrplay_apiService_rsp: ReleaseAll: Setting mKillCmdHandler
Dec 21 09:27:14 raspberrypi sdrplay_apiService: [1470]: devIdx0: sdrplay_apiService_rsp: mHrtbtHandlerThread: mHrtbtHandlerThread: Exit
Dec 21 09:27:14 raspberrypi sdrplay_apiService: [1471]: sdrplay_apiService_device: mCmdHandlerThread: Exit
Dec 21 09:27:14 raspberrypi sdrplay_apiService: [519]: devIdx0: sdrplay_apiService_rsp: ReleaseAll: mCmdHandler killed
Dec 21 09:27:14 raspberrypi sdrplay_apiService: [519]: sdrplay_apiService: heartBeatThread: MSG: DevIdx0: RELEASED

fventuri, 2021-12-21:
@paulh002 - good catch

I just noticed in your code that you set the gain to 88 (see here: https://github.com/paulh002/SoapySDRTest/blob/master/SoapySDRTest/SdrDevice.cpp#L542).
In reality gains (or better gain reductions) work differently on the SDRplay RSPs; you can read about them in the SDRplay API Specification guide (of course, feel free to ask if you have questions about gains).
Also this discussion thread (https://github.com/pothosware/SoapySDRPlay3/issues/35) might give you an idea about the different options we are exploring on how to handle those gains.

Franco

paulh002, 2021-12-21:
Thats it!, just removed the setGain from my test program and than it works. I will check the discussion thread you shared. Hopefully I can implement a generic way to set the gain. The idea of using SoapySDR is to be independend of SDR implementations. But thanks a lot for quick response.

closed 2021-12-21 by paulh002