pothosware/SoapySDR #337: Unable to build on Raspberry Pi 4

patja, 2021-12-31:
I think perhaps there is an issue with where Python libraries are expected to be.

This is on a clean Raspberry Pi 4, following the build instructions from the wiki:

```
-- Build type not specified: defaulting to release.
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
--
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
-- SWIG_FOUND: TRUE - 4.0.2
-- PYTHONINTERP_FOUND: TRUE - 3.9.2
-- PYTHON_EXECUTABLE: /usr/bin/python
Traceback (most recent call last):
  File "/home/pi/SoapySDR/python/get_python_lib.py", line 4, in <module>
    from distutils.sysconfig import get_python_lib
ModuleNotFoundError: No module named 'distutils.sysconfig'
-- PYTHON_INSTALL_DIR: ${prefix}/
-- PYTHONLIBS_FOUND: TRUE - 3.9.2
-- PYTHON_INCLUDE_DIRS: /usr/include/python3.9
-- PYTHON_LIBRARIES: /usr/lib/arm-linux-gnueabihf/libpython3.9.so
-- CMAKE_SWIG_FLAGS=-c++;-threads;-I/home/pi/SoapySDR/include;-DSIZE_T_IS_UNSIGNED_INT
CMake Warning (dev) at /usr/share/cmake-3.18/Modules/UseSWIG.cmake:634 (message):
  Policy CMP0078 is not set: UseSWIG generates standard target names.  Run
  "cmake --help-policy CMP0078" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

Call Stack (most recent call first):
  python/CMakeLists.txt:160 (SWIG_ADD_LIBRARY)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.18/Modules/UseSWIG.cmake:486 (message):
  Policy CMP0086 is not set: UseSWIG honors SWIG_MODULE_NAME via -module
  flag.  Run "cmake --help-policy CMP0086" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/UseSWIG.cmake:736 (SWIG_ADD_SOURCE_TO_MODULE)
  python/CMakeLists.txt:160 (SWIG_ADD_LIBRARY)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at python/CMakeLists.txt:173 (install):
  install TARGETS given no LIBRARY DESTINATION for module target "_SoapySDR".

CMake Error at python/CMakeLists.txt:178 (install):
  install FILES given no DESTINATION!

--
-- #############################################
-- ## Begin configuration for Python3 support...
-- #############################################
-- Enabling optional Python3 bindings if possible...
-- SWIG_FOUND: TRUE - 4.0.2
CMake Warning (dev) at /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args`
  (Python3InterpDbg) does not match the name of the calling package
  (Python3Interp).  This can lead to problems in calling code that expects
  `find_package` result variables (e.g., `_FOUND`) to follow a certain
  pattern.
Call Stack (most recent call first):
  python3/FindPython3Interp.cmake:62 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  python3/CMakeLists.txt:16 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find Python3InterpDbg (missing: PYTHON3_DBG_EXECUTABLE)
-- PYTHON3INTERP_FOUND: TRUE
-- PYTHON3_EXECUTABLE: /usr/bin/python3
Traceback (most recent call last):
  File "/home/pi/SoapySDR/python3/../python/get_python_lib.py", line 4, in <module>
    from distutils.sysconfig import get_python_lib
ModuleNotFoundError: No module named 'distutils.sysconfig'
-- PYTHON3_INSTALL_DIR: ${prefix}/
-- Could NOT find Python3Libs (missing: PYTHON3_LIBRARIES PYTHON3_INCLUDE_DIRS)
-- PYTHON3LIBS_FOUND: FALSE
-- PYTHON3_INCLUDE_DIRS: PYTHON3_INCLUDE_DIR-NOTFOUND
-- PYTHON3_LIBRARIES: PYTHON3_LIBRARY-NOTFOUND
--
-- ######################################################
-- ## SoapySDR enabled features
-- ######################################################
--
 * Library, runtime library v0.8.1-g6f97389b
 * Apps, command line applications
 * Tests, library unit tests
 * Python, python bindings v3.9.2

-- ######################################################
-- ## SoapySDR disabled features
-- ######################################################
--
 * Docs, doxygen documentation
 * Python3, python3 bindings

-- SoapySDR version: v0.8.1-g6f97389b
-- ABI/so version: v0.8
-- Install prefix: /usr/local
-- Configuring incomplete, errors occurred!
See also "/home/pi/SoapySDR/build/CMakeFiles/CMakeOutput.log".

```

guruofquality, 2021-12-31:
>ModuleNotFoundError: No module named 'distutils.sysconfig'

Can you try to install distutils module for python? Its just trying to find out where to install python bindings

patja, 2022-01-01:
I ended up getting it to build.  Later I was thrashing a bit trying different SDR projects and I suspect one of their installations satisfied the missing dependency that was causing the SoapySDR build to fail.  One thing I discovered was missing was pip, perhaps installing that brought along the distutils.sysconfig?

closed 2022-11-24 by ncorgan