pothosware/SoapySDR #338: Python library abends when connecting to AirspyHF device

patja, 2022-01-01:
I have SoapySDR and the AirspyHF driver Installed on a Raspberry Pi 4 running minimal Raspbian.

I am trying to get the example script from https://github.com/pothosware/SoapySDR/wiki/PythonSupport to run.

The Python library is not connecting to the AirspyHF.  Here is what is emitted:

```
{driver=airspyhf, label=AirSpy HF+ [c85282aa96ca3cf8], serial=c85282aa96ca3cf8}
Traceback (most recent call last):
  File "/home/pi/soaptest.py", line 12, in <module>
    sdr = SoapySDR.Device(args)
  File "/usr/local/lib/python3.9/dist-packages/SoapySDR.py", line 1758, in __new__
    return cls.make(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/SoapySDR.py", line 1374, in make
    return _SoapySDR.Device_make(*args)
RuntimeError: SoapySDR::Device::make() no match
```

SoapySDRUtil reports the following.  It seems to see the AIrspyHF OK:

```
pi@raspberrypi:~ $ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-g6f97389b
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libairspyhfSupport.so (0.2.0-d682533)
Available factories... airspyhf
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

pi@raspberrypi:~ $ SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=AirspyHF
  hardware=AirspyHF
  serial=c85282aa96ca3cf8

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CF32, CS16, CS8, CU16, CU8
  Native format: CF32 [full-scale=1]
  Antennas: RX
  Full gain range: [0, 54] dB
    LNA gain range: [0, 6, 6] dB
    RF gain range: [-48, 0, 6] dB
  Full freq range: [0.009, 31], [60, 260] MHz
    RF freq range: [0.009, 31], [60, 260] MHz
  Sample rates: 0.912, 0.768, 0.456, 0.384, 0.256, 0.192 MSps

```

Maybe this is an issue for the AirspyHF plugin/driver?

guruofquality, 2022-01-01:
It could be the python module is linked against another version of SoapySDR also installed.

What search paths does your python module report? Use the same interpreter you used for the script

`path/to/python/interp -c "import SoapySDR; print(SoapySDR.listSearchPaths())"`

patja, 2022-01-01:
Thanks for the suggestion.  I am using python3 to run the script, which is python3.9 on the pi.  All the pythons report the same result:
```

pi@raspberrypi:~ $ python3 -c "import SoapySDR; print(SoapySDR.listSearchPaths())"
('/usr/local/lib/SoapySDR/modules0.8',)
pi@raspberrypi:~ $ python -c "import SoapySDR; print(SoapySDR.listSearchPaths())"
('/usr/local/lib/SoapySDR/modules0.8',)
pi@raspberrypi:~ $ python2 -c "import SoapySDR; print(SoapySDR.listSearchPaths())"
('/usr/local/lib/SoapySDR/modules0.8',)
pi@raspberrypi:~ $ /usr/bin/python3.9 -c "import SoapySDR; print(SoapySDR.listSearchPaths())"
('/usr/local/lib/SoapySDR/modules0.8',)

```

guruofquality, 2022-01-01:
I see the airspihf in the example script, it prints:
{driver=airspyhf, label=AirSpy HF+ [c85282aa96ca3cf8], serial=c85282aa96ca3cf8}

Does the example script still say `args = dict(driver="rtlsdr")` in it? Just make it airspyhf

patja, 2022-01-01:
Ah that was it.  Bad oversight on my part adapting the file.  Thanks!

closed 2022-01-01 by patja