pothosware/SoapySDR #344: PlutoSDR in Windows 10 has a long startup delay

robcazzaro, 2022-02-20:
Hello, I just installed Photosware PothosSDR-2021.05.01-vc16-x64 on my Windows 10 machine, to use with my PlutoSDR. I removed the other device DLLs from the modules.08 directory, and I see what I wanted to achieve

```
C:\Work\SDR\rtl_433>soapysdrutil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-PothosSDR-2021.05.01-vc16-x64
API Version: v0.8.0
ABI Version: v0.8
Install root: C:\Program Files\PothosSDR
Search path:  C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/PlutoSDRSupport.dll (0.2.1-ca40ea9)
Available factories... plutosdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

The weird thing is that when I run a --find, it takes 15 seconds to report the PlutoSDR, and so does --probe. The header part is printed immediately, then it waits at least 15 seconds to print the actual device information

```
C:\Work\SDR\rtl_433>soapysdrutil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

WARNING: Unknown parameter '0' in <context>
WARNING: Unknown parameter '23' in <context>
WARNING: Unknown parameter 'v0.23' in <context>
Found device 0
  device = plutosdr
  driver = plutosdr
  uri = usb:1.35.5

C:\Work\SDR\rtl_433>SoapySDRUtil --probe "device=plutosdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device
WARNING: Unknown parameter '0' in <context>
WARNING: Unknown parameter '23' in <context>
WARNING: Unknown parameter 'v0.23' in <context>
WARNING: Unknown parameter '0' in <context>
WARNING: Unknown parameter '23' in <context>
WARNING: Unknown parameter 'v0.23' in <context>

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=PlutoSDR
  hardware=ADALM-PLUTO
  ad9361-phy,model=ad9364
  ad9361-phy,xo_correction=39999868
  backend_version=0.23 (git tag: v0.23  )
  fw_version=v0.34
  hw_model=Analog Devices PlutoSDR Rev.C (Z7010-AD9364)
  hw_model_variant=1
  hw_serial=1044734c9605000df9ff0500deb64f2518
  library_version=0.19 (git tag: 5f5af2e)
  local,kernel=5.4.0-00535-g9c04de11ae53
  uri=local:
  usb,idProduct=b673
  usb,idVendor=0456
  usb,libusb=1.0.24.11584
  usb,product=PlutoSDR (ADALM-PLUTO)
  usb,release=2.0
  usb,serial=1044734c9605000df9ff0500deb64f2518
  usb,vendor=Analog Devices Inc.

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: NO
  Sensors: xadc_temp0, xadc_voltage0, xadc_voltage1, xadc_voltage2, xadc_voltage3, xadc_voltage4, xadc_voltage5, xadc_voltage6, xadc_voltage7, xadc_voltage8, adm1177_current0, adm1177_voltage0, ad9361-phy_temp0, ad9361-phy_voltage2
     * xadc_temp0: 53.276654 C
     * xadc_voltage0 (vccint): 1.013672 V
     * xadc_voltage1 (vccaux): 1.805420 V
     * xadc_voltage2 (vccbram): 1.014404 V
     * xadc_voltage3 (vccpint): 1.009277 V
     * xadc_voltage4 (vccpaux): 1.805420 V
     * xadc_voltage5 (vccoddr): 1.346191 V
     * xadc_voltage6 (vrefp): 1.246582 V
     * xadc_voltage7 (vrefn): 0.000732 V
     * xadc_voltage8: 0.809326 V
     * adm1177_current0:
     * adm1177_voltage0:
     * ad9361-phy_temp0: 34.211000 C
     * ad9361-phy_voltage2: 0.303418 V

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS8, CS12, CS16, CF32
  Native format: CS16 [full-scale=2048]
  Antennas: A_BALANCED
  Full gain range: [0, 73] dB
    PGA gain range: [0, 73] dB
  Full freq range: [70, 6000] MHz
    RF freq range: [70, 6000] MHz
  Sample rates: 0.065105, 1, 2, 3, 4, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 1, 2, 3, 4, 6, 7, 8, 9, 10 MHz

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CS8, CS12, CS16, CF32
  Native format: CS16 [full-scale=32768]
  Antennas: A
  Full gain range: [0, 89] dB
    PGA gain range: [0, 89] dB
  Full freq range: [70, 6000] MHz
    RF freq range: [70, 6000] MHz
  Sample rates: 0.065105, 1, 2, 3, 4, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 1, 2, 3, 4, 6, 7, 8, 9, 10 MHz
```

My laptop is a new Lenovo T14s Gen 2 with a decent i5-1145G7 and 16Mb RAM, so shouldn't be the limiting factor

Is there a way to reduce the lag? Anything I can do to help troubleshoot? When I use rtl_433 with SoapySDR support, it takes a good 20 seconds to start scanning.

thanks

zuckschwerdt, 2022-02-21:
Since you already narrowed it down to one driver, the time is likely spent in SoapyPlutoSDR.

But I don't see a easy way to pinpoint the IIO function blocking or timing out.
A cheap way would be to pepper https://github.com/pothosware/SoapyPlutoSDR/blob/master/PlutoSDR_Registration.cpp with debug prints and recompile. Or run with profiling instrumentation like Gprof, but I don't know if there is something for Windows.

A quick first step might be to see which [backend](https://github.com/pothosware/SoapyPlutoSDR/blob/master/PlutoSDR_Registration.cpp#L18) is causing the delay, if it is a scan delay.
There is also [feat-libusb](https://github.com/pothosware/SoapyPlutoSDR/blob/feat-libusb/PlutoSDR_Registration.cpp) to accept only the known ADALM-Pluto USB VID:PID (0456:b673) when enumerating USB, which might be a fix.

I guess all troubleshooting needs you to recompile the module though.

zuckschwerdt, 2022-02-21:
Or for some triage: Does it take the same time if there is no Pluto connected? What if there are no driver modules? How about only another driver module, say librtlsdrSupport.so?

robcazzaro, 2022-02-21:
Yes, it takes as long to time out when there's no Pluto SDR attached (sorry, didn't notice it before). I also tried iio_info -s, since on another (Ubuntu) laptop I had problems with internal sensors, but it's not the case. Output of iio_info -s with Pluto connected and Soapysdrutil --find without Pluto is below (both --find and --probe, both have a 15 seconds delay)

I have not yet looked into building on Windows (I did rebuild the library on Ubuntu). As soon as I figure out the right environment, I will try to add debug prints

```
C:\Work\SDR\rtl_433>iio_info -s
Library version: 0.23 (git tag: c14a0f8)
Compiled with backends: xml ip usb serial
Unable to create Local IIO context : Function not implemented (40)
Available contexts:
        0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=1044734c9605000df9ff0500deb64f2518 [usb:1.24.5]
        1: 192.168.2.1 (Analog Devices PlutoSDR Rev.C (Z7010-AD9364)), serial=1044734c9605000df9ff0500deb64f2518 [ip:pluto.local]

C:\Work\SDR\rtl_433>SoapySDRUtil --probe "device=plutosdr,uri=ip:192.168.2.1"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device
←[1m←[31m[ERROR] no device context found.←[0m
Error probing device: no device context found

C:\Work\SDR\rtl_433>SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

No devices found!
```

zuckschwerdt, 2022-02-22:
Just to be sure, what if you move the `PlutoSDRSupport.dll` to another directory, how long does it take then?

robcazzaro, 2022-02-22:
I'm embarrassed to admit that neither PlutoSDR nor Soapy had anything to do with my problem. I wanted to test how long it took when I removed PlutoSDRSupport.dll, and I did it late at night. Surprisingly, it was as fast as expected, no delay.

After a few tests it turns out that playing audio via a USB-connected amplifier introduces a delay. It's not enough to have the USB audio device connected and powered on, something must be playing, too. Not sure why, and something I will need to figure out, but not here :)

Thanks so much for the help and patience. Keep up the great work!

closed 2022-02-22 by robcazzaro

zuckschwerdt, 2022-02-22:
No worries, it's interesting and not something we encountered before. I guess pothosware/SoapyPlutoSDR/pull/50 will fix it. The IIO lib is just to greedy trying to open every USB device available ;) and that might very well timeout for a blocked USB sound card.

zuckschwerdt, 2022-02-22:
If you have a Linux laptop which also exhibits a delay, then testing feat-libusb would be great!
I never encountered USB problems, even when using multiple SDR at the same machine, but I hope that change fixes things.

That branch should fix the assumed USB delay, it will not fix the "other IIO sensors" problem (which I gather should be with the local backend only?).

robcazzaro, 2022-02-22:
Yes, the other IIO sensor is due to local backends. I worked around it by disabling local devices in the iio library (as per one of the suggestions in the issue). It's suboptimal, but works for now. And, no, that laptop doesn't have the USB delay, so won't help, I guess. If I manage to somehow repro, I'll recompile with the pull request and send my findings

No worries, I got a working solution for both machines. Once again, thanks!