pothosware/SoapySDR #345: Unable to find devices on SoapyRemote on WSL2

FenixC4, 2022-03-06:
I have configured my Windows 11 system to have a SoapySDRServer setup on Windows system and SoapySDR with SoapyRemoteModule built on WSL2. When I start the Server I am able to connect to it from Windows system itself and other deviecs in the network, however I am unable to connect to it from WSL2

[image]

[image]

[image]

What should I do to make it work ?

zuckschwerdt, 2022-03-06:
Does `SoapySDRUtil --info` on WSL show the driver?

FenixC4, 2022-03-06:
I'm using `./SoapySDRUtil` . because I have it installed in custom directory, I have done the same on my Ubuntu Laptop and it works just fine.

[image]

And as you can see it shows the driver

zuckschwerdt, 2022-03-06:
Would have been too easy, right ;)
Next guess: with WSL2 the subsystem is a VM, the network interface is mapped to the host (Windows), but since it's external to the host it might need to pass through the firewall?
If you want to cut the remote client out, try `socat`. On WSL a `socat - tcp:192.168.1.109:1234` should connect and the SoapySDRServer should show that connection.

FenixC4, 2022-03-06:
Ok, so I went down the rabbit hole and resolved the issue. It was all down to Windows's firewall.
First I found out that by default Windows doesn't allow WSL to connect to it in any way, so that event pings won't reach Windows system. 

By using this command `New-NetFirewallRule -DisplayName "WSL" -Direction Inbound  -InterfaceAlias "vEthernet (WSL)"  -Action Allow` you allow connections to be made from WSL to Windows machine. Since it only uses the HyperV adapter it is safe and closed from outside connections.

Next I had to add rules both Inbound and Outbound for the SoapyRemote itself since by default it got somehow blocked. After adding these rules everything works just fine.

closed 2022-03-06 by FenixC4