pothosware/SoapySDR #349: Add suport for signal hound

gaur-yogesh, 2022-03-23:
Please add support for signal hound SDR such as BB60c etc

zuckschwerdt, 2022-03-23:
Please show links to Signal Hound API documentation and provide an overview how that API will map to (most) of the functions here: https://github.com/pothosware/SoapySDR/blob/master/include/SoapySDR/Device.h

ncorgan, 2022-03-28:
This has been on the wishlist for a few years, we just don't/didn't have the hardware: https://github.com/pothosware/SoapySDR/issues/185

I remember from looking into it, it's [pretty convoluted](https://signalhound.com/software/signal-hound-software-development-kit-sdk/). A bunch of device-specific APIs with a mish-mash of OS support between them.

closed 2022-03-28 by zuckschwerdt

ncorgan, 2022-12-26:
@gaur-yogesh https://github.com/pothosware/SoapySDR/issues/185#issuecomment-1365121377