pothosware/SoapySDR #351: Teature request: Support for Perseus SDR

nmaster2042, 2022-04-10:
This nice SDR has a lib that could be used to make a SoapySDR driver:
https://github.com/Microtelecom/libperseus-sdr

It would bring access to this hardware from many apps based on Soapy.

ncorgan, 2022-04-17:
It's a pretty simple API. I can do this if someone loans me the hardware.

nmaster2042, 2023-08-14:
SDR++ has source now for perseus sdr, works fine, see:

https://github.com/AlexandreRouma/SDRPlusPlus/tree/master/source_modules/perseus_source

But as you stated, iwth hardware it's better for developing the driver

ncorgan, 2023-08-14:
I already have a local first attempt from some months back, but I’d rather test with hardware before putting it out.

https://github.com/AlexandreRouma/SDRPlusPlus/tree/master/source_modules/perseus_source

nmaster2042, 2023-08-14:
OK, I hope you'll be able to make tests with hardware.

When you're ready, you can let me know, I'll be happy to make a try.