pothosware/SoapySDR #360: Eventual CMake compatibility break

ncorgan, 2022-06-14:
From a recent Windows CI build (CMake 3.23.2):
```
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
CMake Deprecation Warning at swig/python/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.
  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
````

This line is declaring CMake 2.8 as the default. 2.8.8 is from 2012, so even though we list C++11 as the default, we're approaching a possible breaking point.

zuckschwerdt, 2022-06-14:
There is a short compatibility overview on https://gitlab.kitware.com/cmake/community/-/wikis/CMake-Versions-on-Linux-Distros
Requiring CMake 3 will bump the OS minimum to e.g. Debian 8 Jessie and Ubuntu 16.04 Xenial.
(I'd say it's reasonable to drop support for stock Wheezy / Trusty.)

eabase, 2025-02-13:
CMake is ~3.30 today.

ryandesign, 2025-08-30:
It's no longer eventual; master now actually fails to build with cmake 4:

```
CMake Error at swig/python/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!
```

commit 1667b4e6 (2025-08-30): Update swig for compat with newer CMake (fixes #360)
closed 2025-08-30 by zuckschwerdt