pothosware/SoapySDR #370: Module found but No factories found!

stefanino-ch, 2022-07-26:
Raspberry Pi Bullseye 64 bit, all up to date
Soapy itself installed via apt:
```
dpkg --get-selections | grep soapy
libsoapysdr-dev:arm64				install
libsoapysdr0.7:arm64				install
soapyosmo-common0.7:arm64			install
soapysdr-tools					install
```

Download and run SDRPlay API 0.7 64 bit

Download, compile and install SoapySDR:
```
git clone --depth 1 --branch soapy-sdr-0.7.2 https://github.com/pothosware/SoapySDR.git
cd SoapySDR
mkdir build
cd build
cmake ..
make -j4
sudo make install
sudo ldconfig
```
At this point no modules and no factories was found with
SoapySDRUtil --info

Added a symlink: 
```
cd /usr/local/lib
mkdir SoapySDR
cd SoapySDR
sudo ln -s /usr/local/lib modules0.7
```

Now modules are found, but no factories!
```
SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.2-2
API Version: v0.7.1
ABI Version: v0.7
Install root: /usr
Search path:  /usr/lib/aarch64-linux-gnu/SoapySDR/modules0.7       (missing)
Search path:  /usr/local/lib/aarch64-linux-gnu/SoapySDR/modules0.7 (missing)
Search path:  /usr/local/lib/SoapySDR/modules0.7
Module found: /usr/local/lib/SoapySDR/modules0.7/libSoapySDR.so
Module found: /usr/local/lib/SoapySDR/modules0.7/libsdrplay_api.so
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

Now I'm lost, are there any advises how to fix the "No factories found issue"?

zuckschwerdt, 2022-07-26:
> libsoapysdr-dev:arm64				install
> libsoapysdr0.7:arm64				install
> soapysdr-tools					install
> ...
> git clone --depth 1 --branch soapy-sdr-0.7.2 https://github.com/pothosware/SoapySDR.git
> ...
> sudo make install

Now you have SoapySDR libs and util installed twice, to /usr and /usr/local with slightly different settings (at least the runtime/module path).
Those are nearly the same version, but mismatched headers and libs will happen. It's a common source of errors (when the versions are different). Best to choose just one, or if you really want to, use `cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr ..` to clobber the Debian install.

> At this point no modules and no factories was found with SoapySDRUtil --info

Try `/usr/bin/SoapySDRUtil --info` and compare.

stefanino-ch, 2022-07-27:
**Thank you**, based on your comments I was able to solve the issue. 

**Basically two bad things happened:** 

I tried to downgrade from latest soapy to 0.7.2. During this process I uninstalled SoapySDR but left behind the SoapySDRPlay stuff. Downloaded recompiled SoapySDR but not SoapySDRPlay. So in fact I had SoapySDR 0.7.2 and a SoapySDRPlay compiled previously against the latest SoapySDR version. Here the problem was definitely between chair and keyboard. 

Later on during the whole setup of another application (which needs all the Soapy stuff) one of the packages (gr-osmosdr) did a parallel installation of Soapy0.7.2 based on the package dependencies.  This is what was mentioned above. 

**How I solved it:** 

1. Brand new installation of my Raspberry from scratch. 
2. Installed the SDRPlay API
3. Download/ compile SoapySDR 0.7.2
4. Download/ compile SoapySDRPlay latest Version
5. Installed via apt all the other packages needed for my application. This brought in again (because the dependencies) the second SoapySDR but everything is still working as expected. 

**What can maybe improved on my installation (NOT TESTED YET, JUST THINKING!):** 
Install first all packages needed by my application. This will bring in already SoapySDR with the correct version. Compile SoapySDRPlay against this installation. This should avoid having SoapySDR installed twice on different locations.

closed 2022-07-28 by zuckschwerdt