pothosware/SoapySDR #374: Receive / plot samples

snowformatics, 2022-08-28:
Hi,

thanks for building and maintaining the SoapySDR Python API! I am new to SDR and I tried the example code which seems to work:

> print(sr.ret) #num samples or error code
    print(sr.flags) #flags set by receive operation
    print(sr.timeNs) #timestamp for receive buffer

Returns 

> 1024
32
0

 I am using a RSPdx, so my first question is whether the output is as expetced. Second question, I have no idea how to extract the data to generate a spectrum / waterfall diagram. I am good with Python but new to radio. So any help would be very much appreciated.

Thanks a lot
S

fventuri, 2022-08-28:
@snowformatics - Since you are familiar with Python but new to radio, here are a couple of good resources to get you started with your RSPdx, SoapySDR, and Python (and DSP, Digital Signal Processing):

- [Calling all lovers of Python and SDRs](https://www.sdrplay.com/calling-all-lovers-of-python-and-sdrs/) by SDRplay
- [PySDR: A Guide to SDR and DSP using Python](https://pysdr.org/) by Dr. Mark Lichtman

Have fun with your SDR and Python!
Franco

snowformatics, 2022-08-29:
Thanks Franco, this was very helpful! Looking forward to receive some meteor signals!