pothosware/SoapySDR #376: Streaming format scale

fsphil, 2022-10-01:
Just a quick question to clear up some confusion I have on stream formats.

When using CS16 format for example, can I safely assume the values -32768 - 32767 will work?
Or must I call `getNativeStreamFormat()` to see if the native format is also CS16 but the `fullScale` is something different?

Asking as the BladeRF driver expects values in a range of -2048 - 2047 when using CS16, and I've never been sure if that's correct or a bug. Both ways make sense from different perspectives.

zuckschwerdt, 2022-10-01:
The scale is only normalized (to MSB) on non-native streams. The native streams might have a different scale.
Or the the other way around: you need to check/apply `fullScale` but on non-native streams that info is not available and you can assume the full width.

zuckschwerdt, 2022-10-01:
Oh, and one more trap to look out for: there is no consensus if e.g. 12-bit full scale should be 2047 or 2048, so you need to test for both if you want a quick normalization by a shift-left-4.

fsphil, 2022-10-02:
Gotcha. So I always need to check the native format, and if it matches what I'm sending then I have to honour fullScale. That makes sense.

Thanks!

closed 2022-10-02 by fsphil