pothosware/SoapySDR #377: reading settings option list (getSettingInfo) fails using C if no option names set

ahopper, 2022-10-03:
if the settings optionNames is not populated this
https://github.com/pothosware/SoapySDR/blob/f8c17833126397af933237babbcaa14925259eb6/lib/TypeHelpers.hpp#L140
resets the numOptions to 0 so the C api cannot see the options.

the lime driver has an example of not setting optionNames

https://github.com/myriadrf/LimeSuite/blob/e829d3ede6222c1c0fc0e90dcd2050e50569e33d/SoapyLMS7/Settings.cpp#L802-L807

zuckschwerdt, 2022-10-03:
Just swappig the `toStrArray()` call order for `out.options` and `out.optionNames` should work right?
With `options` last it will take precedence over the result from `optionNames` -- a comment to the intention is then also needed ;)
Do you want to PR this change?

ahopper, 2022-10-03:
Yep that should work, I'm happy to create a PR.

closed 2022-10-03 by zuckschwerdt

ericek111, 2024-11-16:
Now, if the option names are not supplied (as they are optional), the `optionNames` list is not properly initialized, causing a crash upon reading the junk `char*`s. :D 

Honestly, I'd just shove `size_t numOptionNames` into the `SoapySDRArgInfo` struct, but that would break ABI. Another option is to pre-fill the names with NULLs (`sizeof(char*) * out.numOptions`, as returned for the options list):

```diff
-static inline char **toStrArray(const std::vector<std::string> &strs, size_t *length)
-{
-    auto out = callocArrayType<char *>(strs.size());
+static inline char **toStrArray(const std::vector<std::string> &strs, size_t *length, size_t minLength = 0)
+{
+    auto out = callocArrayType<char *>(std::max(minLength, strs.size()));
```

What do you think?

zuckschwerdt, 2024-11-17:
Good point and a nice fix indeed. PR if you like.
Please include e.g. `…size of the options and optionNames set, …`
at https://github.com/pothosware/SoapySDR/blob/master/include/SoapySDR/Types.h#L94
and `When null or emtpy, the option value…`
at https://github.com/pothosware/SoapySDR/blob/master/include/SoapySDR/Types.h#L105

guruofquality, 2024-11-17:
Another way would be to copy options into optionNames when the `info.optionNames` is empty. Any app looking for a displayable name for the option would just use the option string itself if optionNames was empty.