pothosware/SoapySDR #381: brew install pothossoapy breaking in GitHub's macos-11 machines

BatchDrake, 2022-11-02:
Hi,

This is an issue I've already opened in `homebrew-pothos` (see https://github.com/pothosware/homebrew-pothos/issues/59), but since I got no answer I thought I was opening it in the wrong repository.  I am copying the description here, feel free to close the one that does not fit in the right place.

> I've been failing to build pothossoapy in macOS 11 for some time now. The build machines are GitHub Action's `macos-11` ones. These are the commands I've been using:
> 
> ```
> $  brew tap pothosware/homebrew-pothos && brew update
> $  brew install pothossoapy
> ```
> 
> And these are the errors I get:
> 
> ```
> Run brew install pothossoapy
>   brew install pothossoapy
>   shell: /bin/bash -e {0}
>   env:
>     BUILD_TYPE: Release
>     pythonLocation: /Users/runner/hostedtoolcache/Python/3.9.14/x64
>     PKG_CONFIG_PATH: /Users/runner/work/SigDigger/Qt/5.14.[2](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:2)/clang_64/lib/pkgconfig
>     Qt5_Dir: /Users/runner/work/SigDigger/Qt/5.14.2/clang_64
>     Qt5_DIR: /Users/runner/work/SigDigger/Qt/5.14.2/clang_64
>     QT_PLUGIN_PATH: /Users/runner/work/SigDigger/Qt/5.14.2/clang_64/plugins
>     QML2_IMPORT_PATH: /Users/runner/work/SigDigger/Qt/5.14.2/clang_64/qml
> ==> Tapping nlohmann/json
> Cloning into '/usr/local/Homebrew/Library/Taps/nlohmann/homebrew-json'...
> Tapped 1 formula (15 files, 45.4MB).
> ==> Downloading https://ghcr.io/v2/homebrew/core/ca-certificates/manifests/2022-10-11
> ==> Downloading https://ghcr.io/v2/homebrew/core/ca-certificates/blobs/sha256:1b264e579e[3](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:3)1b30[4](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:4)1a87ff91f09d[5](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:5)f7cc0d51fea1c83e[6](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:6)3aee1[7](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:7)a1b95509cbe1
> ==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:1b264e579e31b3041a[8](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:8)7ff[9](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:9)1f09d5f7cc0d51fea1c83e63aee17a1b95509cbe1?se=2022-[10](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:10)-12T10%3A30%3A00Z&sig=tOgae%2FZVW7f7%2Fy3feiMUcWPpUD7TkarLe4xlr1XwSBY%3D&sp=r&spr=https&sr=b&sv=2019-12-12
> ==> Downloading https://ghcr.io/v2/homebrew/core/openssl/3/manifests/3.0.6
> ==> Downloading https://ghcr.io/v2/homebrew/core/openssl/3/blobs/sha256:6756c6165f41cb16d5fb738293b7131fab13906ecb73e055efde257[11](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:11)cac95a4
> ==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:6756c6165f41cb16d5fb738293b7131fab13906ecb73e055efde25711cac95a4?se=2022-10-[12](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:13)T10%3A30%3A00Z&sig=ILcX2hjamUplg%2FhCPicH1z4LM8VHHOTNkEKgvB5qdM8%3D&sp=r&spr=https&sr=b&sv=2019-12-12
> ==> Downloading https://ghcr.io/v2/homebrew/core/poco/manifests/1.12.2
> ==> Downloading https://ghcr.io/v2/homebrew/core/poco/blobs/sha256:2d82e850db4860e7e343e6c9a7[13](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:14)408b098615603c62c287c5f0ea5780ea2fc7
> ==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:2d82e850db4860e7e343e6c9a713408b098615603c62c287c5f0ea5780ea2fc7?se=2022-10-12T10%3A30%3A00Z&sig=8bhGrCZZNud%2BGizFoqoPzvfS7feBtKf5zYPS7Re4UG4%3D&sp=r&spr=https&sr=b&sv=2019-12-12
> Warning: nlohmann/json/nlohmann_json has been deprecated because it formula is now part of core; use 'nlohmann-json'!
> ==> Downloading https://github.com/nlohmann/json/archive/v3.10.0.tar.gz
> ==> Downloading from https://codeload.github.com/nlohmann/json/tar.gz/refs/tags/v3.10.0
> ==> Downloading https://github.com/beltoforion/muparserx/archive/v4.0.7.tar.gz
> ==> Downloading from https://codeload.github.com/beltoforion/muparserx/tar.gz/refs/tags/v4.0.7
> ==> Downloading https://github.com/pothosware/PothosCore/archive/pothos-0.7.1.tar.gz
> ==> Downloading from https://codeload.github.com/pothosware/PothosCore/tar.gz/refs/tags/pothos-0.7.1
> ==> Downloading https://ghcr.io/v2/homebrew/core/openssl/1.1/manifests/1.1.1r
> ==> Downloading https://ghcr.io/v2/homebrew/core/openssl/1.1/blobs/sha256:5eeaf8bb4b01748057ca4a923ee580f779d125ac89ee01e6b8656da91deaeed7
> ==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:5eeaf8bb4b01748057ca4a923ee580f779d125ac89ee01e6b8656da91deaeed7?se=2022-10-12T10%3A30%3A00Z&sig=lI%2FT24QevW9f1QKj9GxdTR%2FYjxWeUE0mpm6l5IBwk3A%3D&sp=r&spr=https&sr=b&sv=2019-12-12
> ==> Downloading https://ghcr.io/v2/homebrew/core/readline/manifests/8.2.1
> ==> Downloading https://ghcr.io/v2/homebrew/core/readline/blobs/sha256:e6dfc7d95895f18657c0fb15e77a8c104362bb87bafdff770a6a352301cc1082
> ==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:e6dfc7d95895f18657c0fb15e77a8c104362bb87bafdff770a6a352301cc1082?se=2022-10-12T10%3A30%3A00Z&sig=woKqYza9lJqTGqaKL0EiYdFz98LWpwECRKM6R2X8nA8%3D&sp=r&spr=https&sr=b&sv=2019-12-12
> ==> Downloading https://ghcr.io/v2/homebrew/core/soapysdr/manifests/0.8.1_1
> ==> Downloading https://ghcr.io/v2/homebrew/core/soapysdr/blobs/sha256:12198c0ad0281bf725073d507a78b2ac0b4d588d5106c0cbbbac0165641341ad
> ==> Downloading from https://pkg-containers.githubusercontent.com/ghcr1/blobs/sha256:12198c0ad0281bf725073d507a78b2ac0b4d588d5106c0cbbbac0165641341ad?se=2022-10-12T10%3A30%3A00Z&sig=cjy2AaB3Wd2TqmAZq2sXGvoIGq2q7oXzV8kGFb4cpKo%3D&sp=r&spr=https&sr=b&sv=2019-12-12
> ==> Downloading https://github.com/pothosware/PothosSoapy/archive/pothos-soapy-0.5.1.tar.gz
> ==> Downloading from https://codeload.github.com/pothosware/PothosSoapy/tar.gz/refs/tags/pothos-soapy-0.5.1
> ==> Installing pothossoapy from pothosware/pothos
> ==> Installing dependencies for pothosware/pothos/pothossoapy: ca-certificates, openssl@3, poco, nlohmann/json/nlohmann_json, pothosware/pothos/muparserx, pothos, openssl@1.1, readline and soapysdr
> ==> Installing pothosware/pothos/pothossoapy dependency: ca-certificates
> ==> Pouring ca-certificates--2022-10-11.all.bottle.tar.gz
> ==> Regenerating CA certificate bundle from keychain, this may take a while...
> 🍺  /usr/local/Cellar/ca-certificates/2022-10-11: 3 files, 225.5KB
> ==> Installing pothosware/pothos/pothossoapy dependency: openssl@3
> ==> Pouring openssl@3--3.0.6.big_sur.bottle.tar.gz
> 🍺  /usr/local/Cellar/openssl@3/3.0.6: 6,452 files, 28.4MB
> ==> Installing pothosware/pothos/pothossoapy dependency: poco
> ==> Pouring poco--1.12.2.big_sur.bottle.tar.gz
> 🍺  /usr/local/Cellar/poco/1.12.2: 916 files, 13.8MB
> ==> Installing pothosware/pothos/pothossoapy dependency: nlohmann/json/nlohmann_json
> 🍺  /usr/local/Cellar/nlohmann_json/3.10.0: 6 files, 1.3MB, built in 11 seconds
> ==> Installing pothosware/pothos/pothossoapy dependency: pothosware/pothos/muparserx
[... 7 lines cut ...]
>         DatagramSocket(SocketImpl* pImpl);
>         ^
> /usr/local/include/Poco/Net/DatagramSocket.h:35:2: note: candidate constructor not viable: requires 0 arguments, but 1 was provided
>         DatagramSocket();
>         ^
> /usr/local/include/Poco/Net/DatagramSocket.h:51:2: note: candidate constructor not viable: requires at least 2 arguments, but 1 was provided
>         DatagramSocket(const SocketAddress& address, bool reuseAddress, bool reusePort = false, bool ipV6Only = false);
>         ^
> [  4%] Building CXX object lib/CMakeFiles/Pothos.dir/Version.cpp.o
> cd /tmp/pothos-20221012-24590-1ees939/PothosCore-pothos-0.7.1/build/lib && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DPOCO_ENABLE_CPP11 -DPOCO_ENABLE_CPP[14](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:15) -DPOCO_HAVE_IPv6 -DPOCO_NO_STAT64 -DPOCO_OS_FAMILY_UNIX -DPOCO_UNBUNDLED -DPOTHOS_DLL_EXPORTS -DPOTHOS_LIBRARY_NAME=\"libPothos.0.7.1.dylib\" -I/tmp/pothos-[20](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:21)2[21](https://github.com/Batch [... cut ...]
> Do not report this issue to Homebrew/brew or Homebrew/core!
> 
> These open issues may also help:
> brew soapysdr *before* or *after* brew tap pothosware/homebrew-pothos? https://github.com/pothosware/homebrew-pothos/issues/[58](https://github.com/BatchDrake/SigDigger/actions/runs/3233842836/jobs/5296213541#step:11:59)
> ```
> 
> Since the linked issue provided by brew does not seem to be related to what I am experiencing, I decided to open a new issue here. What could I do?

Thanks in advance,

ncorgan, 2022-11-24:
Looks to be a duplicate of: https://github.com/pothosware/PothosCore/issues/241

ncorgan, 2022-11-29:
Closing as duplicate of https://github.com/pothosware/PothosCore/issues/241

closed 2022-11-29 by ncorgan