pothosware/SoapySDR #383: Airspy Mini with SoapySDR in Ubuntu: seems to work but does not show strong stations in spectra

Kimmo1, 2022-11-02:
Hi

I have installed SoapySDR in Ubuntu 20.04.5 LTS, and I am using it with Airspy Mini.
In principle it seems to work, I can get samples from Airspy, but the spectrum at the frequencies of strong FM stations shows no stations, just noise. The output of the command 'SoapySDRUtil --probe' is given at the end of this mail, and it seems to be OK.

The Python program I have used is the following:

> import SoapySDR
> from SoapySDR import *
> import numpy as np
>
> 
> results = SoapySDR.Device.enumerate()
> args = results[0]
> print("Using driver: %s" % args["driver"])
> sdr = SoapySDR.Device(args)
> 
> sampling_rate = 3e6
> freq = 90e6
> Nsamples = 1024
> 
> \# configure device
> sdr.setSampleRate(SOAPY_SDR_RX, 0, sampling_rate)
> sdr.setFrequency(SOAPY_SDR_RX, 0, 'RF', freq)
> 
> \# set the gains
> sdr.setGain(SOAPY_SDR_RX, 0, 5)
> \# or
> \#sdr.setGain(SOAPY_SDR_RX, 0, 'IF', 5)
> \#sdr.setGain(SOAPY_SDR_RX, 0, 'LNA', 5)
> \#sdr.setGain(SOAPY_SDR_RX, 0, 'MIX', 5)
> 
> \# turn off AGC
> sdr.setGainMode(SOAPY_SDR_RX, 0, False)
> 
> \#setup a stream (complex floats)
> buff = np.array([0]*Nsamples, np.complex64)
> rxStream = sdr.setupStream(SOAPY_SDR_RX, SOAPY_SDR_CF32)
> sdr.activateStream(rxStream) #start streaming
> sr = sdr.readStream(rxStream, [buff], len(buff))
> 
> \#shutdown the stream
> sdr.deactivateStream(rxStream) #stop streaming
> sdr.closeStream(rxStream)

How can I fix this problem ?
cheers, Kimmo 

-------------------------------- output of the command 'SoapySDRUtil --probe'  -------------------------

> \######################################################
> \##     Soapy SDR -- the SDR abstraction library     ##
> \######################################################
> 
> Probe device
> 
> \----------------------------------------------------
> \-- Device identification
> \----------------------------------------------------
>   driver=Airspy
>   hardware=Airspy
>   serial=a74068c830673493
> 
> \----------------------------------------------------
> \-- Peripheral summary
> \----------------------------------------------------
>   Channels: 1 Rx, 0 Tx
>   Timestamps: NO
>   Other Settings:
>      * Bias tee - Enable the 4.5v DC Bias tee to power SpyVerter / LNA / etc. via antenna connection.
>        [key=biastee, default=false, type=bool]
>      * Bit Pack - Enable packing 4 12-bit samples into 3 16-bit words for 25% less USB trafic.
>        [key=bitpack, default=false, type=bool]
> 
> \----------------------------------------------------
> \-- RX Channel 0
> \----------------------------------------------------
>   Full-duplex: NO
>   Supports AGC: YES
>   Stream formats: CS16, CF32
>   Native format: CS16 [full-scale=32767]
>   Antennas: RX
>   Full gain range: [0, 45] dB
>     LNA gain range: [0, 15] dB
>     MIX gain range: [0, 15] dB
>     VGA gain range: [0, 15] dB
>   Full freq range: [24, 1800] MHz
>     RF freq range: [24, 1800] MHz
>   Sample rates: 6, 3 MSps

Kimmo1, 2022-11-04:
The following extra information may help:

1) The same happens if using SoapySDR with RTL-SDR instead of Airspy Mini; no FM stations visible

2) In Ubuntu I can use 'airspy_rx' to save I/Q data from Airspy into file, and then I can use GNU Radio to read the file and compute the spectra:  the FM stations are visible

3) The reason may be some kind of file conflict. I have GNU Radio, and after installing SoapySDR it complains that 
'ImportError: libSoapySDR.so.0.8: cannot open shared object file: No such file or directory '
 when I am trying to use RTL-SDR as a source. This may actually be my own fault, because after installing SoapySDR there were two 'libSoapySDR.so.0.8' files in the system, and I had to remove the other to make SoapySDR to work correctly. 

4) The attached plot shows imaginary parts of 1024 samples from Airspy, when using SoapySDR in Python. The plot looks strange to me, I was expecting to see random values. 

[image]

cheers, Kimmo

ast, 2022-11-04:
I'm working on an improved driver for the Airspy R2 and mini. It's not quite ready yet, but you can follow the development here.

https://github.com/ast/SoapyAirspy

If your issue is not hardware related, I think my driver will eventually solve it.

Kimmo1, 2022-11-04:
Thank you @ast for the information.  

Does it help if I update the firmware of Airspy Mini ?
The command 'airspy_info' gives the following:

Firmware Version: AirSpy MINI v1.0.0-rc10-0-g946184a 2016-09-19

Is it too old ?

cheers, Kimmo

ast, 2022-11-04:
Hi @Kimmo1, I don't know if it will help your problem, but I will only test my new driver with the latest firmware from Airspy.

Kimmo1, 2022-11-11:
Hi 

I updated the firmware of the Airspy Mini, but that did not help. 

However, I noticed that if I increase the number of samples read from Airspy to a value of 16384, there are strong FM stations visible in the spectrum, but the stations are at wrong frequencies...

A similar thing happens when I use RTL-SDR with pyrtlsdr ( https://github.com/pyrtlsdr/pyrtlsdr ), which is a Python wrapper for librtlsdr. There are FM stations visible when the number of samples read from RTL-SDR is for example 1024, but the stations are at correct frequencies only when the number of samples read is equal or greater than 4096. 

I do not know how this is possible.   I had GNU Radio installed before SoapySDR or librtlsdr, if that matters. 

cheers, Kimmo

austinfong, 2023-01-16:
> The following extra information may help:
>
> 1. The same happens if using SoapySDR with RTL-SDR instead of Airspy Mini; no FM stations visible
> 2. In Ubuntu I can use 'airspy_rx' to save I/Q data from [...]

Hi, Kimmo,

Could you help find and share the file "libSoapySDR.so.0.8-2" with me (brianfong@qq.com)?
I am installing this SoaphSDR to CentOS in my pc. 
When I run command #SoaphySDRUtil --info
It shows below: 
SoaphSDRUtil: error while loading shared libraries: libSoapySDR.so.0.8-2: cannot open shared  ....

It looks this file is missing in my pc.
Thanks a lot for any suggestion.