pothosware/SoapySDR #384: Failing to load modules using python bindings

danny1722, 2022-11-08:
when i run the the following code i get thes errors

```
import SoapySDR
from SoapySDR import * #SOAPY_SDR_ constants

args = dict(driver="sdrplay")
sdr = SoapySDR.Device(args)
```

```
[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspyhfSupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspySupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/bladeRFSupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/HackRFSupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/LMS7Support.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/miriSupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/osmosdrSupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/rtlsdrSupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/sdrPlaySupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/uhdSupport.dll)
  LoadLibrary() failed: The specified module could not be found.

[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/volkConverters.dll)
  LoadLibrary() failed: The specified module could not be found.
```
when running the command line utility it works fine 

```
C:\Program Files\PothosSDR\bin>SoapySDRUtil --make="driver=sdrplay"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Make device driver=sdrplay
[INFO] devIdx: 0
[INFO] hwVer: 255
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000
  driver=SDRplay
  hardware=RSP1A
  sdrplay_api_api_version=3.070000
  sdrplay_api_hw_version=255
```
i have already tried reinstalling photos SDR and python 3.9 i also tries adding `os.add_dll_directory("C:\Program Files\PothosSDR\bin")` to the SoapySDR.py file but with no succes. i am using windows 11.

thanks in advance.

gvanem, 2022-11-19:
It's caused by the lines in SoapySDR.i.in:
```
if os.name == 'nt' and hasattr(os, 'add_dll_directory'):
    root_dir = __file__
    for i in range(5): #limit search depth
  ...
```
and the horrid new way Python 3.8+ searches for .DLLs on Windows. 
Ref: https://docs.python.org/3/whatsnew/3.8.html#bpo-36085-whatsnew

gvanem, 2022-11-19:
BTW, this works for me:
```diff
--- a/swig/python/SoapySDR.in.i 2022-02-07 12:19:50
+++ b/swig/python/SoapySDR.in.i 2022-11-19 11:29:16
@@ -28,15 +28,8 @@
 import os

 if os.name == 'nt' and hasattr(os, 'add_dll_directory'):
-    root_dir = __file__
-    for i in range(5): #limit search depth
-        root_dir = os.path.dirname(root_dir)
-        bin_dir = os.path.join(root_dir, 'bin')
-        if os.path.exists(bin_dir):
-            try: os.add_dll_directory(bin_dir)
-            except Exception as ex:
-                print('add_dll_directory(%s): %s'%(bin_dir, ex))
-            break
+    import ntpath
+    os.add_dll_directory (ntpath.realpath(r"..\..\bin"))
 %}
```

danny1722, 2023-01-16:
thanks for the info. i just ended up adding `C:\Program Files\PothosSDR\bin` to system path which solved my issue

closed 2023-01-16 by danny1722