pothosware/SoapySDR #385: our own hardware

nome321, 2022-11-25:
Hello
I am newbie in SoapySDR world 
I am going to implant SoapySDR on our own hardware Nios processor running 
like 
FPGA =  Intel Altera Arria ii gx 
DAC  = AD9148  two channels
ADC =                  two channels 
CONNECTIVITY SFP+
is it possible can we run SoapySDR on our hardware ?
please helping us 

Thanks

zuckschwerdt, 2022-11-25:
> can we run SoapySDR **on** our hardware?

Do you really mean on the Nios? (`Nios -> Soapy -> SFP+ -> Ether/IP? -> GUI software`)
SoapySDR is not a protocol as such, so that would be SoapyRemote then?

I assume you mean "**with** our hardware", right? (`Nios -> SFP+ -> Ether/IP? -> Soapy -> GUI software`)
SFP+ is serious power, very nice. Would the data stream be Ethernet / IP / UDP / TCP?
It shouldn't be too complicated to code a Soapy-module to support your protocol. Do you have any further info on the intended protocol?

nome321, 2022-11-25:
Hello
Thanks for your reply

I assume you mean "with our hardware", right? (Nios -> SFP+ -> Ether/IP? -> Soapy -> GUI software)

Yes This is our Real case 

let me check SoapyRemote if any issue we will face I will share here

Thanks 

Nome

zuckschwerdt, 2022-11-25:
> > I assume you mean "with our hardware", right? (Nios -> SFP+ -> Ether/IP? -> Soapy -> GUI software)

> Yes This is our Real case

That would not be SoapyRemote then. If you have info on your protocol (`SFP+ -> Protocol? -> Soapy`) let us know.

closed 2023-02-22 by zuckschwerdt