pothosware/SoapySDR #386: RX888 MKII 16bit SDR Receiver Radio

PLIW2017, 2022-11-25:
Hi,

I was wondering if there is support planned for the RX888 MKII 16bit SDR Receiver Radio? I’m wanting to use DumpHFDL but via Soapy. Thank you.

fventuri, 2022-11-25:
@PLIW2017 - I saw a few months ago someone posted a repository called SoapyRX888 (https://github.com/cozycactus/SoapyRX888) .

I don't know if it works and if it would work for your use case; perhaps you could get in touch with the author of that module for more information.

Franco

nmaster2042, 2022-11-25:
I asked the author, no reply until now.

fventuri, 2022-11-25:
@nmaster2042 - hmm.. that's a problem then.
Did you try to build it and run it? Did it work for you?

Perhaps the next best place to ask then is Oscar Steila IK1XPV repo (https://github.com/ik1xpv/ExtIO_sddc), since most of the software development of drivers and libraries for the RX888 happens there.

Franco

nmaster2042, 2022-11-25:
@fventuri see here: https://github.com/ik1xpv/ExtIO_sddc/issues/217

fventuri, 2022-11-26:
@nmaster2042 - you are ahead of the game!
I just read that issue (and noticed the fact that the last message from Howard was more than six months ago), and it looks like there's no support for Linux yet.

Franco

nmaster2042, 2022-11-26:
@fventuri yes I know.

Probably this SDR will never be mine.

The main feature was the ability to get the full HF band at once.

nmaster2042, 2023-06-05:
Hi.

Some move on https://github.com/cozycactus/SoapyRX888
based on a lib created by same author: https://github.com/cozycactus/librx888

Hope it will continue.

fventuri, 2023-06-05:
@nmaster2042 Not sure if you already have an RX888; if you do, you might be interested in ka9q-radio (https://github.com/ka9q/ka9q-radio); about a month ago I wrote the 'rxx888d' daemon to allow to run the RX888 with that program (see here: https://groups.io/g/NextGenSDRs/topic/ka9q_radio_with_rx888/98876548).

Franco

nmaster2042, 2023-08-26:
@fventuri, thanks for the tip.

I looked at the ka9q-radio but I'm not very excited about this software.

What I would like is to have a more standard support for this SDR to be able to use on app like Gqx, SDR++, Cubic, etc.

That's why a Soapy driver makes sense for me.

I don't have skills to write something like this, so unless such a driver becomes avaliable, I'll don't buy this SDR.

fventuri, 2023-09-08:
@nmaster2042 - I saw this post yesterday on the NextGenSDRs mailing list, and I thought you might be interested: https://groups.io/g/NextGenSDRs/message/1583

Franco

nmaster2042, 2023-09-08:
@fventuri thank you for the tip, I saw this.

I keep on watching this repo, but for now it seems it doesn't support the mkII.

See https://github.com/cozycactus/SoapyRX888/issues/1

But I still think it's the best way to extend support for this hardware and I hope the dev will continue.

fventuri, 2023-09-13:
@nmaster2042 - good news; it looks like someone got their RX888 mkII working with SDR++ and SoapySDR: https://groups.io/g/NextGenSDRs/message/1606 - they mention a patch for librx888 that hopefully will be included in the official version.

Franco

nmaster2042, 2023-09-13:
@fventuri thanks for the update.

A really hope this development will succeed.

cozycactus, 2023-10-05:
hello!
i'm trying to add here support for mk2, i have only rx888.
https://github.com/cozycactus/librx888/tree/dev

fventuri, 2023-10-05:
Thanks Ruslan!
Did you see the patch that Viktor Zhang wrote to make librx888 work on his RX888 mk II: https://pastebin.com/zjRz62sf

Franco