pothosware/SoapySDR #401: SoapySDR Moduls missing error Message

RaspiGui, 2023-05-07:
Hello! I build and install SoapySDR from "Build Guide" and when I try "SoapySDRUtil --find", I get the message that modules are missing:

######################################################

Soapy SDR -- the SDR abstraction library

######################################################

Lib Version: v0.8.1-g6f97389b
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.8
No modules found!
Available factories... No factories found!
...

I use following SDR hardware: RSPduo

Where to get these modules? Is their an Instruktion to install this Modulen?

zuckschwerdt, 2023-05-07:
You need a module for your SDR hardware. They build and install the same as SoapySDR.
E.g. for an RTL-SDR use https://github.com/pothosware/SoapyRTLSDR -- clone or download and unzip, then
```
cd SoapyRTLSDR
mkdir build
cd build
cmake ..
cmake --build .
sudo cmake --build . -- install
```
(using the traditional "make" the last two last two lines can also simply be `make` then `sudo make install`)

closed 2023-05-07 by zuckschwerdt

RaspiGui, 2023-05-08:
Thanks for your fast reply but im an absolute beginner with Raspberry PI and SDR hardware on Raspberry PI.

Please let me know the link to the modules for RSPduo hardware, i dont use the RTLSDR.

Did you mean following code:

cd SoapySDR
mkdir build
cd build
cmake ..
cmake --build
sudo make build
sudo make install

or did i understand this wrong?

zuckschwerdt, 2023-05-08:
No, the commands you posted won't work. But for RSPduo just use https://github.com/pothosware/SoapySDRPlay3
and read the Wiki https://github.com/pothosware/SoapySDRPlay/wiki on how to build.

RaspiGui, 2023-05-08:
Many thanks.

I follow your description and receive after the cmake .. following error message:

-- Build type not specified: defaulting to release.
-- libsdrplay not found.
CMake Error at CMakeLists.txt:16 (message):
  SDRplay development files not found...

-- Configuring incomplete, errors occurred!
See also "/home/langenhan/SoapySDRPlay/build/CMakeFiles/CMakeOutput.log".

It looks like that something is missing. Is their an libary missing?

zuckschwerdt, 2023-05-08:
Dependencies are missing. You need to get SDR Play driver binaries from: http://sdrplay.com/