pothosware/SoapySDR #408: Build issues

atambi802, 2023-05-12:
Hello, 
I am trying to build soapysdr, on ubuntu 22.04. I keep getting build errors on it. 

`/home/[username]/SoapySDR/swig/python/get_python_lib.py:4: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.sysconfig import get_python_lib
Traceback (most recent call last):
  File "/home/[username]/SoapySDR/swig/python/get_python_lib.py", line 4, in <module>
    from distutils.sysconfig import get_python_lib
ModuleNotFoundError: No module named 'distutils.sysconfig'
CMake Error at swig/python/CMakeLists.txt:128 (install):
  install TARGETS given no LIBRARY DESTINATION for module target
  "_SoapySDR3".
Call Stack (most recent call first):
  swig/python/python3/CMakeLists.txt:1 (BUILD_PYTHON_MODULE)
`

atambi802, 2023-05-12:
This is what is generated in my `CMakeError.log`

`Looking for a CSharp compiler failed with the following output:
CMake Error at /usr/share/cmake-3.22/Modules/CMakeDetermineCSharpCompiler.cmake:5 (message):
  C# is currently only supported for Microsoft Visual Studio 2010 and later.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

CMake Error: CMAKE_CSharp_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "/home/[username]/SoapySDR/build/swig/csharp/CMakeFiles/CheckCSharp/CMakeFiles/CMakeOutput.log".

Performing C++ SOURCE FILE Test SIZE_T_IS_UNSIGNED_INT failed with the following output:
Change Dir: /home/[username]/SoapySDR/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_758b9/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_758b9.dir/build.make CMakeFiles/cmTC_758b9.dir/>
gmake[1]: Entering directory '/home/[username]/SoapySDR/build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_758b9.dir/src.cxx.o
/usr/bin/c++ -DSIZE_T_IS_UNSIGNED_INT   -o CMakeFiles/cmTC_758b9.dir/src.cxx.o -c /home/[username]/SoapySDR/build/CMakeFiles/CMakeTmp/src.cxx
/home/[username]/SoapySDR/build/CMakeFiles/CMakeTmp/src.cxx: In function ‘int main()’:
/home/[username]/SoapySDR/build/CMakeFiles/CMakeTmp/src.cxx:4:38: error: cannot convert ‘unsigned int*’ to ‘size_t*’ {aka ‘long unsigned int*’} in initia>
    4 |     size_t *x = (unsigned int *)(NULL);
      |                                      ^
gmake[1]: *** [CMakeFiles/cmTC_758b9.dir/build.make:78: CMakeFiles/cmTC_758b9.dir/src.cxx.o] Error 1
gmake[1]: Leaving directory '/home/[username]/SoapySDR/build/CMakeFiles/CMakeTmp'
gmake: *** [Makefile:127: cmTC_758b9/fast] Error 2

Source file was:

    #include <cstddef>
    int main() {
    size_t *x = (unsigned int *)(NULL);
    return 0; }
`

Patricia-oid, 2023-05-18:
Me too, kept getting the same error, have you solved this problem?

atambi802, 2023-05-24:
So what I did was install some libs. I think it is because the computer might not have it?
` sudo apt-get install -y libpcap-dev`.

atambi802, 2023-05-24:
Also this:
`sudo apt-get install python3-distutils`
`sudo apt-get install build-essential`

breakthrough55, 2023-06-06:
Me too, kept getting the same error, have you solved this problem?

breakthrough55, 2023-06-06:
@guruofquality

guruofquality, 2023-06-06:
So the original posting mentions ubuntu 22.04 and Python 3.12. So I think thats a newer version of python than the system ships with.

So if I am hearing correctly, the code to get the python library install path is busted on newer version of python because of distutils. Can you share the complete output from running cmake and the version of python on the system?

A quick work around would be to print the desired path for python installs in swig/python/get_python_lib.py

breakthrough55, 2023-06-07:
My configuration is Ubuntu18.04 and Python3.7.  I find  "C# is currently only supported for Microsoft Visual Studio 2010 and later." in CMakeError.log.I think this is a problem with the Cmake version，so i update cmkae version(ubuntu18.04 with Cmake 3.12)to 3.23 . Then I do it again . It works.

closed 2026-10-10 by zuckschwerdt