pothosware/SoapySDR #411: miri driver reports lost samples and lots of "OOOOO" with SoapySDRUtil

peterbmarks, 2023-05-14:
I'm trying to use the miri driver for an SDRPlay clone. Clients, such as CubicSDR have trouble with it and even SoapySDRUtil shows the problem.

SoapySDRUtil --make=driver=miri
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Make device driver=miri
Using device #0: Mirics MSi2500 default (e.g. VTX3D card)
  driver=miri
  hardware=miri
_mirisdr_alloc_async_buffers
Lost samples!
82 01 00 00 00 90 00 00 40 7e 22 fb 0e fc db 4f 
Lost samples!
Lost samples!
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOLost samples!

Does anyone have these devices working?

zuckschwerdt, 2023-05-14:
You cross-posted this to a SoapyMiri issue, let's keep the discussion there. It's not a SoapySDR issue and supporting unlicensed clones isn't well received either.

closed 2023-05-14 by zuckschwerdt

peterbmarks, 2023-05-14:
Fair enough. Thanks.