pothosware/SoapySDR #415: cant find airspy

israelgoldstein, 2023-06-06:
SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-gbb33b2d2
API Version: v0.8.200
ABI Version: v0.8-3
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8-3 (missing)
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

i cant find my airspy, 
what am i missing?

zuckschwerdt, 2023-06-06:
You need modules, see e.g. #401

Note the path is `/usr/local/lib/SoapySDR/modules0.8-3` I would have thought it to be `/usr/local/lib/SoapySDR/modules0.8`?

@ncorgan Is this possibly a bug with some recent build system change?

Psynosaur, 2023-06-11:
I had to compile the modules we wanted to use with the latest ABI version 0.8-3 of SoapySDR to get them to work.

For my use case with C# bindings this was the only way since ABI v0.8 does not have the C# bindings. 

When doing so make sure to set the SOAPY_SDR_ROOT environment variable to the install CMAKE_INSTALL_PREFIX value.

You could also check my github repo :
https://github.com/Psynosaur/SoapySDR-FFT-GUI/tree/main/SoapySDRFFTGUI/Resources/SoapySDR/lib/SoapySDR/modules0.8-3

Airspy is in there

ncorgan, 2023-07-29:
@zuckschwerdt As you saw with the different paths, modules install in per-directory locations for safety. ABI bumps are sometimes common on master (as opposed to stable releases), so this will come up on occasion.

closed 2023-09-07 by zuckschwerdt

eabase, 2025-02-13:
@Psynosaur 
I see you have a bunch of other binaries in the:
https://github.com/Psynosaur/SoapySDR-Spectrum-Analyzer/tree/main/Resources/SoapySDR/bin

Did you build SoapySDR with default settings?