pothosware/SoapySDR #419: PlutoSDR needs zeroconf to work

sensille, 2023-08-06:
I tried to get a PlutoPlus to work via ethernet inside a VM without multicast routing, so auto-discovery doesn't work. adi-osc works when I explicitly give the IP address of the device. I tried the same in gqrx, using the device string "soapy,driver=plutosdr,uri=ip:192.168.1.120", to no avail. Is that supposed to work?

zuckschwerdt, 2023-08-06:
Auto scanned backends are `"local", "usb=0456:b673", "ip"` ([`iio_create_scan_context()`](https://analogdevicesinc.github.io/libiio/v0.16/group__Scan.html#gad40746585b3f8a77e5209301a9ad21cd)) and a fixed address can be give with `hostname` ([`iio_create_network_context()`](https://analogdevicesinc.github.io/libiio/v0.4/group__Context.html#gabb89575442d19f28b6557eee97f8fafd)). The argument `uri` is used for matching only. https://github.com/pothosware/SoapyPlutoSDR/blob/master/PlutoSDR_Registration.cpp

Try the device string `soapy,driver=plutosdr,hostname=192.168.1.120`

closed 2023-08-06 by zuckschwerdt

sensille, 2023-08-06:
Works, thanks!