pothosware/SoapySDR #423: "soapy=0/1/2/3/4..." definitions?

gvimlag, 2023-09-17:
"github.com/pothosware/SoapyRemote/wiki#remote-device-args"
does not explain:
soapy=0
soapy=1
soapy=2
soapy=3
soapy=4
...
Where can I find an explanation of these different "soapy=" values?
Thank you.

zuckschwerdt, 2023-09-17:
It's farther down the page https://github.com/pothosware/SoapyRemote/wiki#grosmosdr-remote-devices

> `soapy` Instructs GrOsmoSDR to use the Soapy support. Required if GrOsmoSDR might find other devices on the local machine.

closed 2023-09-17 by zuckschwerdt

gvimlag, 2023-09-17:
"It's farther down the page https://github.com/pothosware/SoapyRemote/wiki#grosmosdr-remote-devices"

That reference only specifies "soapy=0", and it does not state why it works (no definition).

In other locations around the internet, I see "soapy=1" or "soapy=4", but no explanations as to why?
That's why I'm asking for documentation on:
soapy=0
soapy=1
soapy=2
soapy=3
soapy=4
...

zuckschwerdt, 2023-09-17:
Best I know it's used not in SoapySDR but in GrOsmoSDR, and it's just an enumeration index see e.g. https://github.com/osmocom/gr-osmosdr/blob/master/lib/soapy/soapy_source_c.cc#L118

gvimlag, 2023-09-18:
Does "just an enumeration index" mean that for most SoapySDR cases, "soapy=0" is the same as "soapy=4"?