pothosware/SoapySDR #425: SoapySDR not finding airspy after updates to raspberry pi OS

onigiri63, 2023-10-30:
I've been working on a long-term project that uses the airspyHF+ connected to raspberry pi, using the SoapySDR python library to capture the rf and track signals.  The library was working fine, however I needed to rebuild the pi recently and after running the installation scripts (https://gist.github.com/songritk/b0ac5cde818d42d61a9848f3f2a5a797), SoapySDR refuses to find the airspy.  I know that the airspy is connected and working, since GQRX connects to it fine and is able to see and demodulate radio, however SoapySRD just refuses to acknowledge that the airspy is connected.  I've tried:

-soft and hard reboots
-swapping to different USB ports
-$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 1d50:60a1 OpenMoko, Inc. Airspy
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
-SoapySDRUtil --probe="driver=airspy"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=airspy
Error probing device: SoapySDR::Device::make() no match

I can't find a solution to this issue, it's strange that it was previously working using the installation script I linked, however now it never finds the device.  I'm wondering if there was a change in the Raspbian OS that was pulled in an update, which led to this issue, however right now I'm totally out of ideas on what to do here.

zuckschwerdt, 2023-10-30:
What's the output of `SoapySDRUtil --info`

onigiri63, 2023-10-30:
$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-gbb33b2d2
API Version: v0.8.200
ABI Version: v0.8-3
Install root: /usr/local
Search path:  /usr/local/lib/arm-linux-gnueabihf/SoapySDR/modules0.8-3
Module found: /usr/local/lib/arm-linux-gnueabihf/SoapySDR/modules0.8-3/librtlsdrSupport.so (0.3.3-068aa77)
Available factories... rtlsdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

zuckschwerdt, 2023-10-30:
You only have the SoapyRTLSDR module (librtlsdr) installed. You need to install the https://github.com/pothosware/SoapyAirspyHF module.

closed 2023-10-30 by zuckschwerdt

onigiri63, 2023-10-30:
Okay, I installed the SoapyAirspyHF module, and tried querying both before and after a reboot, however this is what I'm getting:
'''
$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-gbb33b2d2
API Version: v0.8.200
ABI Version: v0.8-3
Install root: /usr/local
Search path:  /usr/local/lib/arm-linux-gnueabihf/SoapySDR/modules0.8-3
Module found: /usr/local/lib/arm-linux-gnueabihf/SoapySDR/modules0.8-3/libairspyhfSupport.so (0.2.0-a2fd6cf)
Module found: /usr/local/lib/arm-linux-gnueabihf/SoapySDR/modules0.8-3/librtlsdrSupport.so   (0.3.3-068aa77)
Available factories... airspyhf, rtlsdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
pi@raspberrypi:~ $ SoapySDRUtil --probe="driver=airspy"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=airspy
Error probing device: SoapySDR::Device::make() no match
'''

zuckschwerdt, 2023-10-30:
Don't use a driver key and it will auto match.
`SoapySDRUtil --probe`
Otherwise use the proper driver key:
`SoapySDRUtil --probe="driver=airspyhf"`

onigiri63, 2023-10-30:
Do you happen to know of a better install script for SoapySDR python on the pi?  Here's what I get from that:

SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
Error probing device: SoapySDR::Device::make() no driver specified and no enumeration results

zuckschwerdt, 2023-10-30:
Sorry, no idea. The SoapyAirspyHF has the ability to automatically find a device, not sure why it doesn't work. https://github.com/pothosware/SoapyAirspyHF/blob/master/Registration.cpp#L43