pothosware/SoapySDR #431: Soapy SDR  No devices found

Oliver0804, 2024-03-28:
```
$SoapySDRUtil --find
Password:
Sorry, try again.
Password:
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################
No devices found! 
```

```
$hackrf_info
hackrf_info version: 2024.02.1
libhackrf version: 2024.02.1 (0.9)
Found HackRF
Index: 0
Serial number: -=-=-=-=-=-=-=
Board ID Number: 2 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: -=-=-=-=-=-=-=
Hardware Revision: older than r6
Hardware supported by installed firmware:
    HackRF One
```

I use macOS with an M3 CPU model. I can successfully identify the HackRF One in GQRX, but I can't find my device using `SoapySDRUtil --find`. What steps might I be missing?

zuckschwerdt, 2024-03-28:
`SoapySDRUtil --info`

Oliver0804, 2024-03-28:
> `SoapySDRUtil --info`

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-release
API Version: v0.8.0
ABI Version: v0.8
Install root: /opt/homebrew
Search path:  /opt/homebrew/lib/SoapySDR/modules0.8
Module found: /opt/homebrew/lib/SoapySDR/modules0.8/librtlsdrSupport.so (0.3.3)
Available factories... rtlsdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

zuckschwerdt, 2024-03-28:
You don't have a hackrf module installed. It should be in `/opt/homebrew/lib/SoapySDR/modules0.8`

Oliver0804, 2024-03-28:
> You don't have a hackrf module installed. It should be in `/opt/homebrew/lib/SoapySDR/modules0.8`

[image]

I only see librtlsdrSupport.so. What should I do next, or what else should I install?

Thank you.

zuckschwerdt, 2024-03-28:
You need this https://github.com/pothosware/SoapyHackRF
No idea if that is available as brew package.

zuckschwerdt, 2024-03-28:
Following the [wiki](https://github.com/pothosware/SoapyHackRF/wiki) it [should be](https://github.com/pothosware/homebrew-pothos/wiki).

Oliver0804, 2024-03-28:
> You need this https://github.com/pothosware/SoapyHackRF No idea if that is available as brew package.

 I will try to compile and install it. Thank you.

Oliver0804, 2024-03-28:
> 維基[百科](https://github.com/pothosware/SoapyHackRF/wiki)[應該是](https://github.com/pothosware/homebrew-pothos/wiki)的。
```
SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 
  version = 2024.02.1
```

Success! Thank you so much.

closed 2024-03-28 by Oliver0804

ybm911, 2024-04-19:
Please note here that if you are using brew-installed Soapy, the compiled and installed SoapyHackRF will appear in the wrong directory:
```
/usr/local/lib/SoapySDR/modules0.8/libHackRFSupport.so
```
You need to specify the file location during compilation:
```
cmake -DCMAKE_INSTALL_PREFIX=/opt/homebrew ..
```
Or copy it after compilation.
This issue is covered here: https://github.com/pothosware/SoapyHackRF/issues/42#issue-1407991082