pothosware/SoapySDR #433: SoapyAirspy crashes when making device with log level SOAPY_SDR_DEBUG

gmbertani, 2024-04-04:
Hi all,

my application (http://www.gabb.it/echoes/) crashes silently under Windows10 when opening an Airspy device.

After some debugging, I've seen the exception causing the crash:

```
+++++++++
Debug Assertion Failed!

Program: ...nts\sviluppo\echoes-related\echoes-git\trunk\debug\echoes.exe
File: minkernel\crts\ucrt\inc\corecrt_internal_stdio_output.h
Line: 2571

Expression: ("Invalid integer length modifier", 0)

For information on how your program can cause an assertion
failure, see the Visual C++ documentation on asserts.

(Press Retry to debug the application)
++++++++

```
The problem originates in Settings.cpp line 65:
`SoapySDR_logf(SOAPY_SDR_DEBUG, "Found AirSpy device: serial = %16Lx", serial);`

logf() calls implicitly std::vsnprintf() that dislikes the "%16Lx" specifier. 

To fix the bug I replaced that line with:

`SoapySDR_logf(SOAPY_SDR_DEBUG, "Found AirSpy device: serial = %" PRIx64, serial);  `

and included inttypes.h 

Note that the problem was highlighted in Echoes because the `SoapySDR_setLogLevel(SOAPY_SDR_DEBUG) `function is called at program's startup, otherwise it wouldn't be noticeable.

Sincerely,
Giuseppe Massimo Bertani

zuckschwerdt, 2024-04-04:
Good find. The modifier should have been at least `ll` (afaics `L` is for floating point?), but `PRIx64` is more portable.

Do you want to PR that to https://github.com/pothosware/SoapyAirspy ? Othwise I can just commit a change and still give you attribution.

gmbertani, 2024-04-05:
Thanks. 
Yes, I've seen that `ll` works too but, as you said, `PRIx64` is more portable. 
I have never made changes to other people's projects on github so I wouldn't want to find myself causing any damage. I guess I should commit Setting.cpp followed by push, right?

zuckschwerdt, 2024-04-05:
The steps would be: fork, clone your fork, add a branch, change files, commit, push, open a PR. Wait for merge then delete the fork. Try if you like, no chance to damage anything.

gmbertani, 2024-04-05:
uhm... thanks but it looks a big deal for 2 lines of code.  Feel free to make the change yourself when you like. 
Cordially 
Giuseppe

closed 2024-04-05 by zuckschwerdt